freifunk-gluon/gluon #3174: Netgear WAX220 100mbit/s ethernet no link up

Djfe, 2024-01-28:
Freifunk Aachen received a report about a Netgear WAX220 not being able to connect to the internet via Devolo PowerLAN while a direct ethernet cable worked.

We got their device to test and figured out that it's not able to get a working link to 100mbit/s port. We verified this with a Fritz!Box 3272.
1gbit/s ports worked (Wan led on the WAX220 also lit up), while 100mbit/s ports didn't (led off, ip a shows "down, no-carrier" etc.)

We need to collect further evidence still but wanted to let others know early.
We are certain the 2.5gbit/s realtek phy and it's driver support are at fault here.

rotanid, 2024-04-21:
@Djfe is this still the case based on current openwrt 23.05? if yes, has it been tested with openwrt master?

rotanid, 2024-12-23:
closing stale issue

closed 2024-12-23 by rotanid