freifunk-gluon/gluon #3181: hiveap's don't survive autoupdate to new releases (from v2019/v2021)

Djfe, 2024-02-04:
Aachen lost 100% of their Hive APs during our migration.
5x HiveAP 121
2x HiveAP 330

the 330s were lost on update from v2019 to v2021.
They are both installed at the same spot but we only managed to find one so far.
A powercycle was enough to bring it back to live.
We lost it again a couple days later on the update from v2021 to v2023.

The 121s updated just fine from v2019 to v2021 but we're lost a couple days later, when we rolled out v2023 for them.

We haven't really gotten hands on the devices, yet.
But we wanted to let people know about this issue :)
If we get hands on, then we'll retest our migration and watch logs and we will also test v2023.2

Djfe, 2024-02-04:
Update:
330 got a major raise in compat version on OpenWrt release 22
https://github.com/openwrt/openwrt/commit/cfe79f2eb8b3c874f3d1a4441b8a60af32169974

I have no clue why they went offline then unless they weren't able to reactivate wifi (for meshing) after trying to install the sysupgrade. (which probably gets tried every hour)

kpanic23, 2024-02-04:
Apparently it's a bug in the autoupdater: The Image check fails after all network has stopped, but network isn't started again after the failure, so the device is left without any network connectivity.
https://github.com/freifunk-gluon/gluon/issues/2742#issuecomment-1780898705

rotanid, 2024-02-04:
the actual issue is described in #2742 as @kpanic23 already wrote. you have to act manually on site.
@neocturne do you think the autoupdate might really be an issue here like @kpanic23 says? at the least the devices should continue to run with the old firmware...

rotanid, 2024-02-04:
hm, this issue is simply not true as they "survive" it other than stated.
and if the issue with the 121 is the same, we don't need two issues for the same thing.
rather a new one about the autoupdater - if it actually is a bug there.

regarding known issues: you can answer it yourself by looking at the past release notes ;-)
at least the issue is marked as a known issue on github since 2023-10-27

kpanic23, 2024-02-05:
> ... at the least the devices should continue to run with the old firmware...
Technically they do. The problem is that the autoupdater stops networking before the update, aborts because of image version mismatch and just exits without re-enabling networking. This results in the node fully running but without any network connectivity. I've removed the image from our update server, so the nodes don't get stuck every hour again trying to upgrade.
If this gets fixed in autoupdater, the fix should probably get backported to 2021.1.x as well, just for safekeeping.

blocktrron, 2024-02-05:
@kpanic23 This was fixed with https://github.com/freifunk-gluon/packages/commit/0648b2dbf71c16127ae498cc14f1d33c3f95c741

rotanid, 2024-02-05:
> @kpanic23 This was fixed with [freifunk-gluon/packages@0648b2d](https://github.com/freifunk-gluon/packages/commit/0648b2dbf71c16127ae498cc14f1d33c3f95c741)

oh wow, and reported by myself over 5 years ago...
somehow we missed to backport it, d'oh!

rotanid, 2024-05-11:
@Djfe can we close this issue?

blocktrron, 2024-07-06:
Closed as not a bug.

closed 2024-07-06 by blocktrron

kpanic23, 2024-07-06:
> the openwrt name changed slightly in ath79 The manifest is correct but the sysupgrade doesn't have the metadata for the old ar71xx name

I just had a look at our firmware archive. It's been "aerohive-hiveap-121" since v2018.2.3, no change at all.