freifunk-gluon/gluon #3183: Xiaomi Mi Router 4A (100M International Edition) is now available in Version 2 that is not supported (device_name = R4ACV2)

ChristianMiddendorf, 2024-02-05:
Hello,
there is now a new version 2 of the inexpensive Xiaomi Mi Router 4A (100M International Edition).
The current Gluon image does not work with this version.
The Xiaomi MiR Patcher can open a Telnet access. The device then reports as device_name = R4ACV2 rom_version = 2.20.21 release.
The challenge is the different flash layout, well documented at https://forum.openwrt.org/t/xiaomi-4a-r4acv2-new-version-bricked/166933/16. There you can already find a customized OpenWRT image as a proof-of-concept.
root@XiaoQiang:~# cat /proc/mtd
dev: size erasesize name
mtd0: 01000000 00010000 "ALL"
mtd1: 00020000 00010000 "Bootloader"
mtd2: 00010000 00010000 "NULL"
mtd3: 00010000 00010000 "Factory"
mtd4: 00010000 00010000 "crash"
mtd5: 00010000 00010000 "cfg_bak"
mtd6: 00100000 00010000 "overlay"
mtd7: 00c60000 00010000 "OS1"
mtd8: 00b00000 00010000 "rootfs"
mtd9: 00230000 00010000 "disk"
mtd10: 00010000 00010000 "Config"
Caution: When debricking via the Xiaomi tool, the device then reports as R4AC while retaining the new layout, which further reduces the distinguishability of the devices. >>>You just see the difference with cat /proc/mtd
Image used: gluon-ffin-v2023.1.1-23+97-l2tp-xiaomi-mi-router-4a-100m-international-edition-sysupgrade.bin of the Freifunk Ingolstadt Community. 
Two suggestions:
- Mark the previous image as valid for V1 (R4AC).
- It would be great if an adaptation and provision for V2 (R4ACV2) were possible.
If required, I can send a device (from Germany) to the person who wants to take care of the issue.

Dexruus, 2024-02-06:
Can confirm that, got a batch of A100 from 09/010 2023 and they even had the flash built in like the flash from the gigabit version in them. Their flashlayout was completely different then the "old ones". Bricked one of my routers on that note already and even the "unbrick" tool didnt work.

Atleast its confirmed now somewhat.

rotanid, 2024-12-23:
the device is support in OpenWrt 24.10 branch since August 2024:
https://git.cdn.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=f51cb7447390949feff598b833b764ac1f3ff508

so you could try to add it to gluon's main branch, too and use this checklist for your contribution:
https://github.com/freifunk-gluon/gluon/wiki/Device-Integration-checklist

ChristianMiddendorf, 2025-01-06:
[ramips-mt76x8.zip](https://github.com/user-attachments/files/18323386/ramips-mt76x8.zip)

The following colums are new:

device('xiaomi_mi-router-4a-100m-intl-v2', 'xiaomi_mi-router-4a-100m-intl-v2', {
	factory = false,
})

ChristianMiddendorf, 2025-01-06:
[gluon-ffin-v2023.2-258+116-fastd-xiaomi_mi-router-4a-100m-intl-v2-sysupgrade.zip](https://github.com/user-attachments/files/18323402/gluon-ffin-v2023.2-258%2B116-fastd-xiaomi_mi-router-4a-100m-intl-v2-sysupgrade.zip)

As a proof of concept the image for the Freifunk Ingolstadt community

commit 715abd7f (2025-03-29): Merge pull request #3414 from ChristianMiddendorf/main

ramips-mt76x8: Add support for Xiaomi Mi Router 4A (100M International Edition v2 - R4ACv2) - closes #3183
closed 2025-03-29 by neocturne