freifunk-gluon/gluon #3246: v2023.2: Status page broken, no IP shown on map

citronalco, 2024-04-09:
### Bug report

**What is the problem?**

APs with Gluon v2023.2 and fastd do not display any IP address on the map.
When opening the AP's status page, an error message is shown:

[image]

Strangely when simply replacing fastd with tunneldigger in the config, both issues are gone.

**What is the expected behaviour?**

IP shown on map, working status page

**Gluon Version:**

v2023.2.2-2-g00160f5

**Site Configuration:**

https://git.bingo-ev.de/freifunk/ffin-site
(master branch, commit a1d7612eaed5d67809abbab674a106bf706a0cd8)

image-customization-fastd.lua for fastd, image-customization-l2tp.lua for L2TP

**Custom patches:**

blocktrron, 2024-04-09:
Please check if respondd is running on the node and send the syslog from an affected node.

citronalco, 2024-04-09:
respondd is running:
```
# ps  | grep respond
 7010 root      2844 S    /usr/bin/respondd -d /usr/lib/respondd -p 1001 -g ff02::2:1001 -i vx_mesh_other -i mesh0 -i mesh-vpn -g ff05::2:1001 -i br-client -t 10
```

Seems like something's wrong with gluon-radv-filterd:

```
# logread
Tue Apr  9 22:05:26 2024 kern.notice kernel: [    0.000000] Linux version 5.15.150 (uid-108@6d2f746b9743) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23715+7-c909fdad18) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Wed Jan 17 23:52:09 2024
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] MIPS: machine is TP-Link TL-WR1043ND v2
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Zone ranges:
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Early memory node ranges
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Tue Apr  9 22:05:26 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Tue Apr  9 22:05:26 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16256
Tue Apr  9 22:05:26 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] Memory: 55980K/65536K available (5661K kernel code, 569K rwdata, 1268K rodata, 1196K init, 211K bss, 9556K reserved, 0K cma-reserved)
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] NR_IRQS: 51
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] CPU clock: 720.000 MHz
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5309056796 ns
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.000001] sched_clock: 32 bits at 360MHz, resolution 2ns, wraps every 5965232126ns
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.008373] Calibrating delay loop... 358.80 BogoMIPS (lpj=1794048)
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.075041] pid_max: default: 32768 minimum: 301
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.080751] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.088546] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.102249] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.112768] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.120057] pinctrl core: initialized pinctrl subsystem
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.127422] NET: Registered PF_NETLINK/PF_ROUTE protocol family
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.134275] thermal_sys: Registered thermal governor 'step_wise'
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.150658] clocksource: Switched to clocksource MIPS
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.163757] NET: Registered PF_INET protocol family
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.169174] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.177778] tcp_listen_portaddr_hash hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.186854] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.195119] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
Tue Apr  9 22:05:26 2024 kern.info kernel: [    0.203291] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[... cut ...]
Tue Apr  9 22:06:10 2024 daemon.info dnsmasq[2691]: using nameserver 2003:f0:671f:52fd:cd:e0ff:fe04:b000#53
Tue Apr  9 22:06:10 2024 daemon.info dnsmasq[2691]: using nameserver 10.42.0.1#53
Tue Apr  9 22:06:18 2024 daemon.err gluon-radv-filterd[4278]: Error loading shared library libnl-tiny.so: No such file or directory (needed by /usr/lib/libbatadv.so)
Tue Apr  9 22:06:18 2024 daemon.info procd: Instance gluon-radv-filterd::instance1 s in a crash loop 10 crashes, 0 seconds since last crash
```

blocktrron, 2024-04-09:
Can you check if `libnl-tiny` is installed? `opkg list-installed | grep libnl`

citronalco, 2024-04-09:
Yes it is:
libnl-tiny1 - 2023-07-27-bc92a280-1

neocturne, 2024-04-10:
Can you provide a download for the image in question?

citronalco, 2024-04-10:
Sure: http://firmware.freifunk-ingolstadt.de/images/latest/experimental_l2tp/sysupgrade/
(scroll down to get to v2023.2)

Currently we have only four APs on v2023.2. Two with L2TP, two with fastd.
My statement above ("Strangely when simply replacing fastd with tunneldigger in the config, both issues are gone.") is wrong: L2TP is also affected, albeit one AP is working.

Broken (fastd):
https://map.freifunk-ingolstadt.de/#!/de/graph/18a6f7cf13ba
https://map.freifunk-ingolstadt.de/#!/de/graph/46a24f6527b7
Broken (l2tp):
https://map.freifunk-ingolstadt.de/#!/de/graph/1459c0cd17a6

Working (l2tp):
https://map.freifunk-ingolstadt.de/#!/de/graph/6466b3ded32c

neocturne, 2024-04-10:
Okay, so the issue is that somehow all the binaries are linked against `libnl-tiny.so`, however the installed file is called `libnl-tiny.so.1`. `libnl-tiny.so.1` is also the SONAME found in `libnl-tiny.so.1`.

I can think of only one explanation for such a thing to happen: There is an outdated `libnl-tiny.so` from an older Gluon version in your `staging_dir`, although it is not clear to me why that would not have been overwritten by the latest build (which should have a symlink `libnl-tiny.so` -> `libnl-tiny.so.1`). In any case, doing a full clean of your build directory (`make dirclean`) should fix the issue.

citronalco, 2024-04-16:
Yes, `make dirclean` fixed it. Sorry for the noise.

closed 2024-04-16 by citronalco