freifunk-gluon/gluon #3262: fix issue that causes dnsmasq spam

Djfe, 2024-05-14:
dnsmasq started spamming my logs a couple months back
```
Wed May 15 01:34:07 2024 daemon.info dnsmasq[1]: reading /tmp/resolv.conf.d/resolv.conf.auto
Wed May 15 01:34:07 2024 daemon.info dnsmasq[1]: using nameserver 2a03:2260:3006::53#53
Wed May 15 01:34:07 2024 daemon.info dnsmasq[1]: using only locally-known addresses for lan
```

this is mostly just a bit of an annoyance because it makes the logs harder to read.
atleast Gluon v2023.2 is affected, v2023.1 might be affected as well (I think I remember seeing it there as well).

I just wanted to report this here so we don't forget about it.
Also the spam could be a symptom of an actual issue (dnsmasq restarting every 5minutes)

site: https://github.com/ffac/site/tree/v2023.2.2-1
gluon tag: 00160f5bd48261156cef1c4e626e1b8dffba2650

maurerle, 2024-06-26:
I heard from @grische that this has been seen on ffdon before their migration to ffmuc:
https://github.com/ffdon/site-ffdon/blob/master/site.conf

As most other communities do not experience this, while I heard that FFAC also experienced this in the old 2019.x firmware (as far as I heard), I looked into common configuration of the two:
https://github.com/ffac/site/blob/v2023.2.x/site.conf

Both firmwares do not have a `dns` section baked in the config.

Most other communities and the new FFMUC config do have this:
https://github.com/freifunkMUC/site-ffm/blob/stable/domains/ffdon_nordwest.conf

one can see, that dnsmasq restart as it rewrites the config: `ls -l --full-time /tmp/resolv.conf.d/resolv.conf.auto`

Adding:
`uci add_list dhcp.@dnsmasq[0].server='2a03:2260:3006::53' && uci commit dhcp`
did not fix the restart. (though it seemed to occur less often..)

Maybe building a firmware with dns settings baked in helps.
It might be that a configuration as "gluon does not know about a dns server as it gets one pushed from the kea from the supernode anyway" does lead to these restarts.
I do not know what it could be other than that.

neocturne, 2024-06-26:
Discussion of the issue on IRC has revealed that it was caused by two supernode radvds announcing different lists of DNS Search Domains, resulting in frequent rewrites of `resolve.conf.auto`.

closed 2024-06-26 by neocturne