freifunk-gluon/gluon #3263: Ubiquiti ERX config mode not indicated by LEDs

Djfe, 2024-05-15:
### Bug report

**What is the problem?**
Ubiquiti Edgerouter-X doesn't blink secondly while in config mode.

**What is the expected behaviour?**
PWR LED should be blinking secondly while in config mode.

**Gluon Version:**
00160f5bd48261156cef1c4e626e1b8dffba2650

**Site Configuration:**
https://github.com/ffac/site/tree/v2023.2.2-1

**Custom patches:**
https://github.com/ffac/site/tree/v2023.2.2-1/patches

Djfe, 2024-06-13:
ups, never replied here.
Thanks a lot for the forum link. it explains a lot. 👍
In this case there's nothing to improve. (OpenWrt can't control any of the led's via software according to the forum)
*close*

Djfe, 2024-06-13:
that would work for the switch led's. (they are controlled by the switch chip)
according to the forum the power led is wired to the power supply. so it is not possible for stock firmware to control it either, is it?

neocturne, 2024-06-14:
> that would work for the switch led's. (they are controlled by the switch chip) according to the forum the power led is wired to the power supply. so it is not possible for stock firmware to control it either, is it?

Yes, I think that is correct.

For the switch, reconfiguring the LEDs as GPIOs is possible in the Device Tree, but I don't think doing so makes much sense.