freifunk-gluon/gluon #3268: v2023.2.x: ramips-mt7620: archer-c20i: config lost

maurerle, 2024-05-30:
### Bug report

**What is the problem?**

When booting an Archer C20i in Config mode and saving the settings, the device boots back into config mode and does not remember any setting on gluon v2023.2.x

Using OpenWrt, setting a config and rebooting does not have problems on OpenWrt 23.05.03 or OpenWrt snapshot (from today).

Affected firmware can be found here:
https://firmware.freifunk-aachen.de/firmware/download/from-2023.2.x/sysupgrade/gluon-ffac-v2023.2.2-1-tp-link-archer-c20i-sysupgrade.bin (logread: https://pastebin.com/m86SjHZk)
and here:
https://firmware.ffmuc.net/next/sysupgrade/gluon-ffmuc-v2024.4.2-next-tp-link-archer-c20i-sysupgrade.bin
(logread at https://pastebin.com/saWP9Y0N
as this was reproducible with FFAC and FFMUC firmware.

I was looking into this issue as the running C20i was bricked by a v2023.2.2 update.
When flashing v2023.2.x on top of a configured v2023.1.x router, the device is stuck in some useless state without wifi and has to be recovered.

**What is the expected behaviour?**

Device should be running fine and save the configuration.

A workaround is to use gluon v2023.1.x based versions.

**Gluon Version:**

```
Hostname:               	ffmuc-14cc2037bdbf
MAC address:            	14:cc:20:37:bd:bf
Hardware model:         	TP-Link Archer C20i
Gluon version / Site version:	v2023.2.2+ / v2022.10.5-next1-132-g83e0b01
Firmware release:       	v2024.4.2-next
Site:                   	Freifunk München
Domain:                 	ffmuc_welt
Public VPN key:         	n/a
Switch type:            	none
```

**Site Configuration:**

FFAC and FFMUC

**Custom patches:**

blocktrron, 2024-05-30:
Your jffs2 partition seems to be to small to retain data

```
Thu Jan 18 00:52:28 2024 kern.notice kernel: [    0.935834] 1 squashfs-split partitions found on MTD device rootfs
Thu Jan 18 00:52:28 2024 kern.notice kernel: [    0.942190] 0x000000770000-0x0000007a0000 : "rootfs_data"
```

blocktrron, 2024-05-30:
THis package includes wpa-supplicant: https://github.com/freifunk-gluon/community-packages/blob/1e70e0744846de83f0b42384bd326a1cde5d9a1a/ffac-autoupdater-wifi-fallback/Makefile#L16

I assume (at least a part) of your size-increase is because of this package.

rotanid, 2024-05-30:
shouldn't such problems be caught during the build process?

blocktrron, 2024-05-30:
@rotanid We sadly don't track this, as we don't know the blocksize of the flash which is the limiting factor :/ It only catches cases, where the image is to big to fit in the flash partition w/o the overlay considered.

blocktrron, 2024-06-02:
@maurerle I've just had a second look. In contrast to our package inclusion, you also include packages for USB Serial / storage / Networking.

Removing them will free enough space to have the affected nodes retain their config again.

I'll close this, as the issue is related to your included package set.

closed 2024-06-02 by blocktrron

rotanid, 2024-06-02:
hm, i already removed additional usb packages for the c20 in February 2023, i assume i had a build warning or something...
https://github.com/tecff/site-ffa/blob/next/site.mk#L418