freifunk-gluon/gluon #3280: Proposal: Rename development branch from master to main

neocturne, 2024-06-11:
As proposed by me in our April meetup:

It would be nice to follow many other projects and rename our development branch to main. When renaming the branch via Github, retargetting all open PRs would be taken care of automatically, and the Github web UI will redirect URLs from the old to the new branch.

The change would not be without churn however: According to https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/managing-branches-in-your-repository/renaming-a-branch, all accesses using Git are *not* redirected, so everyone pulling from the Gluon master would need to update their configuration. Github actions mentioning the master branch would also need to be adjusted.

Options we may want to consider:
- Keep a master branch synced with the main branch for some time like OpenWrt does (automatically via Github actions?)
- Perform the rename shortly after a major Gluon release?

Mentioning all owners of the Gluon org for feedback, as we didn't have enough maintainers present at the last meetup for a decision:
- @AiyionPrime 
- @blocktrron 
- @christf 
- @jplitza 
- @lemoer 
- @mweinelt 
- @rotanid 
- @T-X 

Please react with :+1: or :-1: or comment if you have concerns.

mweinelt, 2024-06-11:
Ideally the master branch would be gone, so that everyone would need to migrate instead of accidentally working on a stale branch.

AiyionPrime, 2024-06-11:
Für mich klingt es als käme GitHub uns da schon gut entgegen:

_Renaming a branch will:

- Re-target any open pull requests
Update any draft releases based on the branch
- Move any branch protection rules that explicitly reference the old name
- Show a notice to repository contributors, maintainers, and admins on the repository homepage with instructions to update local copies of the repository
- Show a notice to contributors who git push to the old branch
- Redirect web requests for the old branch name to the new branch name
- Return a "Moved Permanently" response in API requests for the old branch name_

blocktrron, 2024-06-11:
I don't care about this change, so take me as neutral.

> Perform the rename shortly after a major Gluon release?

I don't think we need to align this, given releases are built from release branches anyways.

However, please check if the CI or other tooling matches on branch naming.

rotanid, 2024-06-11:
neutral, too. so it's decided.
I assume, that the person bringing it up also takes care of the changes? @neocturne

neocturne, 2024-06-13:
- For freifunk-gluon/packages, `master` has been renamed to `main`. To keep old Gluon versions buildable if they reference master explicitly, the current state of the branch has also been pushed as `master `, and made read-only using a branch protection rule
- For freifunk-gluon/gluon, `master` has been renamed to `main`. No master branch exists now; unfortunately, it's not possible to prevent pushing a new master branch by accident, as branch protection rules only apply to existing branches.
- Remaining references to the branch in the main repo are updated in https://github.com/freifunk-gluon/gluon/pull/3282
- Once that is merged, I will also announce the change on the mailing list

herbetom, 2024-06-13:
At least with the newer Rulesets Feature it's possible to prevent creation:

1. https://github.com/freifunk-gluon/gluon/settings/rules
2. New Ruleset -> New branch ruleset
4. give it a name
5. Active enforcement
6. empty bypass actors
7. Add Target -> Include by pattern
8. pattern: master
9. enable "Restrict creations"
10. Create

blocktrron, 2024-07-06:
@neocturne This can be marked as done, right?

closed 2024-07-09 by mweinelt