freifunk-gluon/gluon #3290: DSA interface name overlap: wan1 could be ethernet or private wifi

Djfe, 2024-06-21:
Switching to DSA introduced new interface names like wan1, wan2, lan1, lan2, lan3
This could cause conflicts, once devices like this one end up in Gluon: MQMaker - WiTi Board
`ucidef_set_interfaces_lan_wan "lan1 lan2 lan3 lan4" "wan1 wan2"`
https://github.com/openwrt/openwrt/blob/66edb2f9ad89206ac6f608ec76fad161b53683a7/target/linux/ramips/mt7621/base-files/etc/board.d/02_network#L128

wan1 is currently used if you enable the private wifi feature (for the second band)

If this is an actual conflict, then we should rename the private wifi interfaces eventually.

maurerle, 2024-06-22:
Adding a suffix like _wifi
So that the iface name is wan_wifi0 and wan_wifi1 is enough in this line
https://github.com/freifunk-gluon/gluon/blob/986679a473730bc72b052da52f97f71ab89cce65/package/gluon-web-private-wifi/luasrc/lib/gluon/config-mode/model/admin/privatewifi.lua#L83

Though this is currently not a bug, as no supported device does have such an iface name.
It makes sense though to change this as the proposed name is more descriptive.

neocturne, 2024-06-23:
There's a second piece of code to adjust in `gluon-private-wifi/luasrc/lib/gluon/upgrade/325-gluon-private-wifi`.

It might be a good idea to move interface naming helpers to a central package, so they can be adjusted in a single place if needed.

PR #3294 (2024-06-28): private-wifi: rename ifname to wan_wifiX to not collide with openwrt devices with more than one wan port
Currently, no device is affected, though it makes sense to improve the current situation by giving more meaningful ifnames anyway.

closes #3290
closed 2024-06-28 by blocktrron