freifunk-gluon/gluon #3291: Please add Support for Xiaomi MI Router 4a Gigabit Edition v2

ChristianMiddendorf, 2024-06-23:
Hello,
please add the following section into the file ramips-mt7621 to get the new Xiaomi MI Router 4a Gigabit Edition v2 supported.
It works at least with Gluon V2023.2.3 with the settings for Freifunk Ingolstadt. Completed file based on V2023.2.3 added as .zip.

device('xiaomi-mi-router-4a-gigabit-edition-v2', 'xiaomi_mi-router-4a-gigabit-v2', {
	factory = false,
})

[ramips-mt7621.zip](https://github.com/user-attachments/files/15945319/ramips-mt7621.zip)

herbetom, 2024-06-24:
Sorry, i'm not really willing to open a kinda random zip. It would be best to open a Pull Request.

A mandatory requirement for inclusion is a filled out copy of our checklist:

https://github.com/freifunk-gluon/gluon/wiki/Device-Integration

More information on the whole proccess of adding hardware support can be found here:
- https://gluon.readthedocs.io/en/latest/dev/hardware.html
- https://github.com/freifunk-gluon/gluon/wiki/Device-Integration

If you need further help feel welcome to ask. Either here or via Chat: https://matrix.to/#/#gluon:hackint.org

ChristianMiddendorf, 2024-06-24:
Hello, 
a pull is not allowed or i do not know how to get access. 
[ramips-mt7621.txt](https://github.com/user-attachments/files/15960189/ramips-mt7621.txt)
I attached the ramips-mt7621 file with an ending .txt so it just has to be renamed to ramips-7621 after downloading.
[supported_devices.rst.txt](https://github.com/user-attachments/files/15960234/supported_devices.rst.txt)
The same with the supported_devices.rst file... has to be renamed.
[Device-Integration-Checklist-Xiaomi4aGigabitV2.txt](https://github.com/user-attachments/files/15960261/Device-Integration-Checklist-Xiaomi4aGigabitV2.txt)
Attached you find the Device Integration Checklist as Textfile.
![Mac_Adress_Proof](https://github.com/fre
[image]
ifunk-gluon/gluon/assets/132701936/a47a2585-a3b1-4a25-8458-6ef0ab9e20b1)
Best regards
Christian

ChristianMiddendorf, 2024-06-24:
[image]

grische, 2024-06-25:
@ChristianMiddendorf you first need to create a fork: 
- https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/fork-a-repo

Once you have a fork, you add the changes to the git repository of the fork and commit it:
- https://docs.github.com/en/desktop/adding-and-cloning-repositories/cloning-a-repository-from-github-to-github-desktop

Alternatively, you can also just go into your fork on Github and start editing files there within the web browsers. That's quite handy if you have don't want to deal with the git clones and commits.

Once you have commited to changes (either using the git tools or Github web UI), you can open a pull request:
- https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork

PR #3293 (2024-07-01): ramips-mt7621: add support for Xiaomi Mi Router 4a Gigabit Edition v2
Closes #3291

- [x] Must be flashable from vendor firmware
  - [ ] Web interface
  - [ ] TFTP
  - [x] Other: <First: Set Password for WLAN and System in Original Web interface. Second: Install Exploit as described at OpenWRT, Third: Copy sysupgrade image to /tmp via FTP, Forth: Open Telnet Session to go to /tmp directory to call mtd -e OS1 -r write name-of-your-sysupgrade-image.bin OS1  >
https://openwrt.org/inbox/toh/xiaomi/xiaomi_mi_router_4a_gigabit_edition#notes_on_firmware_exploit_procedure
- [x] Must support upgrade me [... cut]
closed 2024-07-01 by blocktrron