freifunk-gluon/gluon #3298: Please add hardware support for Netgear EX6130 with target ramips-mt7620

fabianh95, 2024-06-30:
- [x] Must be flashable from vendor firmware
  - [x] Web interface
  - [ ] TFTP
  - [x] nmrpflash
- [x] Must support upgrade mechanism
  - [x] Must have working sysupgrade
    - [x] Must keep/forget configuration (`sysupgrade [-n]`, `firstboot`)
  - [x] Gluon profile name matches autoupdater image name
        (`lua -e 'print(require("platform_info").get_image_name())'`)
         netgear-ex6130
- [x] Reset/WPS/... button must return device into config mode
- [x] Primary MAC address should match address on device label (or packaging)
      (https://gluon.readthedocs.io/en/latest/dev/hardware.html#hardware-support-in-packages)
  - When re-adding a device that was supported by an earlier version of Gluon, a
    factory reset must be performed before checking the primary MAC address, as
    the setting from the old version is not reset otherwise.
- Wired network
  - [x] should support all network ports on the device
  - [x] must have correct port assignment (WAN/LAN)
    - if there are multiple ports but no WAN port:
      - the PoE input should be WAN, all other ports LAN
      - otherwise the first port should be declared as WAN, all other ports LAN
- Wireless network (if applicable)
  - [x] Association with AP must be possible on all radios
  - [x] Association with 802.11s mesh must work on all radios 
  - [x] AP+mesh mode must work in parallel on all radios
- LED mapping
  - Power/system LED
    - [x] Lit while the device is on
    - [x] Should display config mode blink sequence 
          (https://gluon.readthedocs.io/en/latest/features/configmode.html)
  - Radio LEDs
    - [no led] Should map to their respective radio
    - [no led] Should show activity
  - Switch port LEDs
    - [no led] Should map to their respective port (or switch, if only one led present) 
    - [no led] Should show link state and activity
- Outdoor devices only:
  - [ ] Added board name to `is_outdoor_device` function in `package/gluon-core/luasrc/usr/lib/lua/gluon/platform.lua`
- Cellular devices only:
  - [ ] Added board name to `is_cellular_device` function in `package/gluon-core/luasrc/usr/lib/lua/gluon/platform.lua`
  - [ ] Added board name with modem setup function `setup_ncm_qmi` to `package/gluon-core/luasrc/lib/gluon/upgrade/250-cellular`
- Docs:
  - [ ] Added Device to `docs/user/supported_devices.rst`

[ex6130.patch](https://github.com/user-attachments/files/16044820/ex6130.patch)

herbetom, 2024-07-02:
Hi @fabianh95,

please open a Pull Request. In the following comment you can find some Infos on how to do that:

https://github.com/freifunk-gluon/gluon/issues/3291#issuecomment-2189307565

The Commit Message should be along the lines of the following:

```
ramips-mt7620: add support for Netgear EX6130
```

If there are problems along the way feel free to ask for help.

fabianh95, 2024-07-04:
Hi @herbetom,

thanks for your answer and sorry for the wrong placed issue.
I created a new [pull request](https://github.com/freifunk-gluon/gluon/pull/3304).

closed 2024-07-06 by blocktrron