freifunk-gluon/gluon #3305: mediatek-filogic: weird tq on wr3000

maurerle, 2024-07-05:
General instability on mediatek filogic devices with mt7915e have been seen, especially on the WR3000, WAX220 and others.
It has to be noted that some devices work better than others. Heavy wifi mesh seems to make the situation worse.

**What is the problem?**

An example of this is this behavior is this device:
https://grafana.ffac.rocks/d/000000002/node?orgId=1&var-node=80afca06d558&from=1718344052951&to=1718403869219&viewPanel=13
[image]

which includes very varying TQ of the device.

The latest finding is this:
https://grafana.ffac.rocks/d/000000002/node?orgId=1&var-node=80afca06d558&from=1720175532350&to=1720193698710&var-select_hostname=ffac-seilpforte-wr3000&var-hostname=ffac-seilpforte-wr3000&var-saveinterval=1m&var-nodetolink=0c0e76cf5d5e&viewPanel=13
[image]

At 1. I restarted the wifi driver using `rmmod mt7915e && modprobe mt7915e`
At 2. I added another mesh device with which this device could mesh on mesh1, creating the timeout issue without the device being possible to reload the firmware
At 3. I restarted the device, as nothing helped.

Afterward, the weird changing TQ can be seen, which behaves in weird waves.

The current workaround includes reloading the mt7915e driver and rebooting the device once the mt7915e bug from #3154 occurs.
A package for this can be found here: https://github.com/ffac/gluon-packages/tree/main/ffac-mt7915-hotfix/files/lib/gluon/mt7915

As @nrbffs also noted on IRC, some other people reported instability with these devices as well. Currently, reloading the wifi driver twice a day seems to help in this situation..

This issue is not about #3154 but about the weird changing TQ leading to bad mesh quality and wifi quality.

**What is the expected behaviour?**

Mesh and wifi quality should be stable on mediatek filogic devices such as the WR3000.

**Further steps**

- [x] check logread to find any related messages (nothing found)
- [ ] test what happens if mesh is using HT20 instead of HE20 (decrease wifi standard from wifi6 to wifi4)
- [ ] check `ls /sys/kernel/debug/ieee80211/phy*/mt76` to find something

### TX_Stats

I found that on other devices `cat /sys/kernel/debug/ieee80211/phy1/mt76/tx_stats` does only show values for 1 to 4 while the affected WR3000 has values for 1 to 8

```
Phy 0, Phy band 0
Length:        1 |   2 - 10 |  11 - 19 |  20 - 28 |  29 - 37 |  38 - 46 |  47 - 55 |  56 - 79 |  80 -103 | 104 -127 | 128 -151 | 152 -175 | 176 -199 | 200 -223 | 224 -247 | 
Count:      6743 |     5177 |      604 |      141 |      145 |        0 |        1 |        0 |        0 |        0 |        0 |        0 |        0 |        0 |        0 | 
BA miss count: 7072

Tx Beamformer applied PPDU counts: iBF: 0, eBF: 2461
Tx Beamformer Rx feedback statistics: All: 541, HE: 539, VHT: 2, HT: 0, BW20, NC: 8286, NR: 8589
Tx Beamformee successful feedback frames: 0
Tx Beamformee feedback triggered counts: 0
Tx multi-user Beamforming counts: 0
Tx multi-user MPDU counts: 0
Tx multi-user successful MPDU counts: 0
Tx single-user successful MPDU counts: 482790

Tx MSDU statistics:
AMSDU pack count of 1 MSDU in TXD:   227714 ( 99%)
AMSDU pack count of 2 MSDU in TXD:      180 (  0%)
AMSDU pack count of 3 MSDU in TXD:       86 (  0%)
AMSDU pack count of 4 MSDU in TXD:       71 (  0%)
AMSDU pack count of 5 MSDU in TXD:       39 (  0%)
AMSDU pack count of 6 MSDU in TXD:       31 (  0%)
AMSDU pack count of 7 MSDU in TXD:       20 (  0%)
AMSDU pack count of 8 MSDU in TXD:      129 (  0%)
```

I do not really know if this is related or not, just a finding.

**Gluon Version:**
v2023.2.3

**Site Configuration:**
[ffac](https://github.com/ffac/site) @ v2023.2.3-2

**Custom patches:**
see site

blocktrron, 2024-07-06:
Can you check if the tx retries / tx failed counters from `iw dev mesh{0,1} station dump` are continously incrementing?

maurerle, 2024-07-06:
They are slightly increasing, but most of the time, they are constant.

## tx failed

```
root@ffac-seilpforte-wr3000:~# iw dev mesh0 station dump | grep "tx failed"
	tx failed:	2228
	tx failed:	47
	tx failed:	2249
	tx failed:	127
	tx failed:	535

# after 10 minutes
root@ffac-seilpforte-wr3000:~# iw dev mesh0 station dump | grep "tx failed"
	tx failed:	2236
	tx failed:	85
	tx failed:	2259
	tx failed:	171
	tx failed:	535
```

## tx retries
```
root@ffac-seilpforte-wr3000:~# iw dev mesh0 station dump | grep "tx retries"
	tx retries:	2223
	tx retries:	47
	tx retries:	2234
	tx retries:	124
	tx retries:	506

# after 10 minutes
root@ffac-seilpforte-wr3000:~# iw dev mesh0 station dump | grep "tx retries"
	tx retries:	2231
	tx retries:	85
	tx retries:	2243
	tx retries:	168
	tx retries:	506
```

batctl p  towards some mesh partner often does not work either, with package losses above 90%.
Does this help?

maurerle, 2024-07-11:
I just tested the MTK patch:
https://github.com/freifunk-gluon/gluon/commit/dd114b5fc2dec3f2e7feef52a7238399b39f0a9e 
from @blocktrron's branch:
https://github.com/freifunk-gluon/gluon/compare/main...blocktrron:gluon:mtk-git-txs.patch

It looked good until I reloaded the driver at about 7:20
Then we had the usual airtime and link stability problems.
Until I reloaded the driver again at 08:45.
Problems then started again at 9:30

[image]

The logread still does not hint to something useful.
So this issue is waiting for other ideas for now :)

T-X, 2024-07-13:
Some driver hiccup is probably the most likely. But still wanted to ask, as it's not clear to me from these graphs alone: Has external traffic causing these losses been ruled out? Is there some available airtime graph? Does it correlate with some route changes in batman-adv?

(I've seen funny route flapping / TQ changes/breakdowns caused by unicast traffic in the past in a test in a specific setup years ago when it was still 802.11g, caused by a hidden node problem: https://www.open-mesh.org/projects/batman-adv/wiki/Bcast-hidden-node. There it would oscillate between the good two-hop route and a bad, direct 1-hop route. Even if CTS/RTS was enabled for unicast traffic. Traffic over the two-hop route would interfere with the batman-adv OGM broadcasts... causing a breakdown in TQ and then switching to 1-hop. Then the TQ would improve, things would switch back to 2-hop. Rinse and repeat. Usually the hidden-node-problem should be quite rare though. Maybe even less likely with newer 802.11 revisions / improvements therein?)

maurerle, 2024-07-15:
Thanks for looking into this @T-X .
In FFAC we only have the WR3000 and WAX220 devices as filogic.
This did not yet occur on ramips-mt7621 devices (which also have the mt7915e wifi driver).

In our test-setup we experienced the same problems with a WAX220:
https://grafana.ffac.rocks/d/000000002/node?orgId=1&var-node=9418654360cb&from=1720773987464&to=1720797801279 (I did not see any correlation with Traffic, gateway, reboot or anything)

There seems to be another test-installation with wr3000 and wax220 which does not have such issues according to its grafana:
https://grafana.ffac.rocks/d/000000002/node?orgId=1&var-node=80afca09d90c&from=1720305372298&to=1721030139574
https://grafana.ffac.rocks/d/000000002/node?refresh=1m&orgId=1&var-node=941865436b4b&from=now-30d&to=now-1m
The TQ has these drops as we reload the mt7915 driver three times a day for filogic hardware due to these connectivity problems.

Remotely, I can only debug by checking the TQ with mesh partners (if there are any) - the actual symptom is that the wifi is unusable as a client when connected to the node during the times in which the device has 3-4% of TQ to neighbors (even though it might have a 100% TQ to mesh-vpn, so it surely is the wifi driver)

I don't think it is a flapping route in batman, though I can not exclude this completely.

maurerle, 2024-09-02:
I just noticed, that before some firmware iterations, the max TQ and min TQ were both fluctuating:

[image]
(see [here - wr3000](https://grafana.ffac.rocks/d/000000002/node?orgId=1&var-node=80afca06d558&from=1718344052951&to=1718403869219&var-hostname=ffac-seilpforte-wr3000&var-saveinterval=1m&var-nodetolink=a0a3f0489b70&viewPanel=7))
another example is [this one - wr3000](https://grafana.ffac.rocks/d/000000002/node?orgId=1&var-node=80afca06d558&var-select_hostname=ffac-seilpforte-wr3000&var-hostname=ffac-seilpforte-wr3000&var-saveinterval=1m&var-nodetolink=0c0e76cf5d5e&from=1718170754843&to=1718756240615&viewPanel=7
):

The latest v2023.2.x firmware does have a solid max tq but still varying min tq which seems broken.
AFAIK the wifi is still unstable for clients - don't know if it got better - I am using the mesh quality as a remote monitoring indicator of the mesh.

The solid max tq can be seen [here - wax220](https://grafana.ffac.rocks/d/000000002/node?orgId=1&var-node=9418654360cb&var-select_hostname=ffac-WAX220&var-hostname=ffac-WAX220&var-saveinterval=1m&var-nodetolink=0c0e76cf5d65&from=1723451416564&to=1724334618769&viewPanel=7):
[image]

### The same bad mesh symptoms were found on the NWA55AXE as well

[image]
this is ramips-mt7621 though ([linkt to grafana - nwa55axe](https://grafana.ffac.rocks/d/000000002/node?refresh=1m&orgId=1&var-node=b8d526ff2996&var-select_hostname=ffac-LustAufLife-Zentrum&var-hostname=ffac-LustAufLife-Zentrum&var-saveinterval=1m&var-nodetolink=b8d526ff8c27&from=now-3h&to=now-1m&viewPanel=7))

## openwrt master

I did build firmware from openwrt master to test, though it sometimes did not even load the wifi driver at all (for the wr3000) and did show the above behavior as well for the WAX220.

So the problem is still not solved in openwrt master as of August 2024 (commit [5d2a008670122f3f69eb3ab4f776d9fe9b6d76dd](https://github.com/openwrt/openwrt/commit/5d2a008670122f3f69eb3ab4f776d9fe9b6d76dd)).

maurerle, 2025-01-08:
I am cautiously optimistic regarding the behavior on current gluon main as the wr3000 seems to have quite stable mesh links for a few hours for about the first time in our mesh setup:

[image]

Will take some time to know if issues really occur :)

maurerle, 2025-01-09:
Die Freude hielt nicht lange an :)

We are back to the old behavior here

[image]

Eventually this is a problem for the NWA50AX Pro only.
Eventually this gets fixed through a proper fix of the basic rates..?
We will wait and see :)

Just a short update to not get people to enthusiastic :laughing:

maurerle, 2025-01-09:
From a debugging session with @blocktrron :

first activate mt76 debug logs:
```bash
cd  /sys/kernel/debug/ieee80211/phy0/mt76
echo 1 > fw_debug_wm
```

Now one can see recovery attempts of the firmware with `logread | grep wsysSerExtCmd` like

```
Thu Jan  9 20:56:20 2025 kern.info kernel: [39867.251830] ieee80211 phy0: WM: (1201.734003:11:SER-W)wsysSerExtCmd: action(3), ser_set(3)
Thu Jan  9 20:57:08 2025 kern.info kernel: [39914.609622] ieee80211 phy0: WM: (1249.091883:24:SER-W)wsysSerExtCmd: action(3), ser_set(3)
```

this can be triggered using sys_recovery:

```
cat  /sys/kernel/debug/ieee80211/phy0/mt76/sys_recovery 
Please echo the correct value ...
0: grab firmware transient SER state
1: trigger system error L1 recovery
2: trigger system error L2 recovery
3: trigger system error L3 rx abort
4: trigger system error L3 tx abort
5: trigger system error L3 tx disable
6: trigger system error L3 bf recovery
7: trigger system error full recovery
8: trigger firmware crash
```

sometimes, it looks like a higher value is needed to recover correctly

writing `echo 7 >  /sys/kernel/debug/ieee80211/phy0/mt76/sys_recovery` (full recovery) has helped to fix the mesh TQ going down for now.
Eventually a lower number is enough as well.

only affected device for now is the zyxel nwa50ax-pro

maurerle, 2025-01-10:
The NWA50AX-Pro did have the issue again this morning:

echoing any lower values than 7 to sys_recover did not have an impact to recovering

[image] 

The issue occured at 07:05
at 07:25 I think `echo 4 > sys_recovery` was used while `echo 6 > sys_recovery` was used at 07:40 ish - though I don't know what made it, as the device tries to recover on its own as well.

While broken, a lot of `muruMlrHandle: can not find empty entry` is spammed to the log.

Echoing 7 did a full recovery and worked:

```
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.051927] ieee80211 phy0: WM: ( 419.476984:99:PM-E)PsCnt[0][1]=1 p=5 BMC_DELIVER_SP=1 BMC_DELIVER_DETECT=44 BmcCnt=0
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.051966] ieee80211 phy0: WM: ( 419.477259:00:PM-E)PsCnt[1][2]=0 p=0 BMC_DELIVER_SP=0 BMC_DELIVER_DETECT=0 BmcCnt=0
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052053] ieee80211 phy0: WM: ( 419.477289:01:PM-E)PsCnt[2][17]=1 p=6 BMC_DELIVER_SP=0 BMC_DELIVER_DETECT=0 BmcCnt=0
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052188] ieee80211 phy0: WM: ( 419.477411:02:PM-E)PsCnt[3][18]=0 p=0 BMC_DELIVER_SP=0 BMC_DELIVER_DETECT=0 BmcCnt=0
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052201] ieee80211 phy0: WM: ( 419.477472:03:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 0] 965 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052209] ieee80211 phy0: WM: ( 419.477503:04:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 1] 238 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052248] ieee80211 phy0: WM: ( 419.477534:05:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 2] 0 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052315] ieee80211 phy0: WM: ( 419.477595:06:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 3] 34 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052349] ieee80211 phy0: WM: ( 419.477625:07:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 4] 5 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052388] ieee80211 phy0: WM: ( 419.477656:08:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 5] 893 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052416] ieee80211 phy0: WM: ( 419.477717:09:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 6] 5 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052449] ieee80211 phy0: WM: ( 419.477747:10:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 7] 19 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052483] ieee80211 phy0: WM: ( 419.477778:11:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 8] 1 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052517] ieee80211 phy0: WM: ( 419.477808:12:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 9] 31 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052552] ieee80211 phy0: WM: ( 419.477839:13:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 10] 0 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052624] ieee80211 phy0: WM: ( 419.477900:14:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 11] 4 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052658] ieee80211 phy0: WM: ( 419.477930:15:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 12] 5 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052692] ieee80211 phy0: WM: ( 419.477991:16:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 13] 5 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052726] ieee80211 phy0: WM: ( 419.478022:17:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 14] 5 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.052761] ieee80211 phy0: WM: ( 419.478052:18:MURU-E)[muruSplMonitorByTimer] NGSplRecord[AC 15] 812 (0 means NG)
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.287178] ieee80211 phy0: WM: ( 419.712427:19:MURU-E)muruMlrHandle: can not find empty entry
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.287390] ieee80211 phy0: WM: ( 419.712671:20:MURU-E)muruMlrHandle: can not find empty entry
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.287549] ieee80211 phy0: WM: ( 419.712824:21:MURU-E)muruMlrHandle: can not find empty entry
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.287713] ieee80211 phy0: WM: ( 419.713007:22:MURU-E)muruMlrHandle: can not find empty entry
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.287890] ieee80211 phy0: WM: ( 419.713190:23:MURU-E)muruMlrHandle: can not find empty entry
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.288003] ieee80211 phy0: WM: ( 419.713282:24:MURU-E)muruMlrHandle: can not find empty entry
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.288195] ieee80211 phy0: WM: ( 419.713465:25:MURU-E)muruMlrHandle: can not find empty entry
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.289012] ieee80211 phy0: WM: ( 419.714136:26:MURU-E)muruMlrHandle: can not find empty entry
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.289653] ieee80211 phy0: WM: ( 419.714930:27:MURU-E)muruMlrHandle: can not find empty entry
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.290354] ieee80211 phy0: WM: ( 419.715632:28:MURU-E)muruMlrHandle: can not find empty entry
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.291015] ieee80211 phy0: WM: ( 419.716303:29:MURU-E)muruMlrHandle: can not find empty entry
Fri Jan 10 07:56:37 2025 kern.info kernel: [79474.291780] ieee80211 phy0: WM: ( 419.717066:30:MURU-E)muruMlrHandle: can not find empty entry
[... cut ...]
Fri Jan 10 07:56:40 2025 daemon.info hostapd: client0: STA 02:6f:82:bc:33:27 IEEE 802.11: disassociated due to inactivity
Fri Jan 10 07:56:41 2025 daemon.info hostapd: client0: STA 02:6f:82:bc:33:27 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jan 10 07:56:59 2025 daemon.info hostapd: client0: STA 8a:f0:1c:52:0e:fb IEEE 802.11: authenticated
Fri Jan 10 07:56:59 2025 daemon.info hostapd: client0: STA 8a:f0:1c:52:0e:fb IEEE 802.11: associated (aid 7)
```

Other filogic devices like WR3000 and WAX220 were not affected yet

maurerle, 2025-01-28:
@blocktrron had the idead to set the radios to HT20 instead of HE20.

This fixes the TQ issue for the WAX220 and the WR3000. The NWA50AX-pro stays barely usable.
I am not sure why this device behaves especially bad, as all 3 filogic devices are installed at the same location.

mean TQ of the NWA50AX-pro to neighbours:
[image]

The max-TQ of this mesh connection is 100%, while the min TQ is about similar to the mean TQ.

maurerle, 2025-08-05:
This issue seems to be fixed with a current build from main.
Investigated TQ for 2 days now, will investigate further, but so far, this looks good.

closed 2025-11-18 by neocturne