freifunk-gluon/gluon #3319: device not leaving config mode: Too few erase blocks

Djfe, 2024-07-18:
### Bug report

**What is the problem?**
logread during config mode shows:
`[  100.108766] Too few erase blocks (1)`
If I save, then it won't save and reenter config mode without any warning whatsoever.

**What is the expected behaviour?**
We could grey out the save button and display a warning message in-case this happens.
This way users can complain and we can build proper firmware in a smaller size.

**Gluon Version:**
gluon-v2023.2.3+

**Site Configuration:**
https://github.com/ffac/site/tree/v2023.2.3-4
Probably irrelevant but it causes the images being too big for the device om2p-lc in particular.
Which we can now fix, but this issue remains.

**Custom patches:**
https://github.com/ffac/site/tree/v2023.2.3-4/patches

rotanid, 2024-07-28:
maybe this will be solved by https://github.com/freifunk-gluon/gluon/pull/3097 ?
@blocktrron am i right or wrong?

Djfe, 2025-01-26:
That will reduce the likelyness of it happening because communities won't create too large images anymore.
Though there are certain devices like the TP-Link CPE series for which #3097 can't verify the size, yet.

Better find a way to tell the user in config mode (warning) that the currently installed image exceeds the possible size and needs to be replaced by a smaller image via a sysupgrade to exit config mode successfully.

rotanid, 2025-07-06:
> [#3097](https://github.com/freifunk-gluon/gluon/pull/3097) has been merged.

but there hasn't been an answer if that fixes this reported problem, was there?

Djfe, 2025-07-07:
imo there's still stuff to do to make the situation  better.
next step for me is to eventually rebuild om2p devices with said PR (or rather main branch) and test, whether it works correctly for them