freifunk-gluon/gluon #3349: ZTE MF289F doesn't finish rebooting sometimes

Djfe, 2024-10-08:
### Bug report

ZTE MF289F (IPQ40XX) on executing reboot doesn't always boot through and then requires a power-cycle.
We have noticed this on two of these devices so far. We haven't debugged this serially

**Gluon Version:**
00160f5bd48261156cef1c4e626e1b8dffba2650

**Site Configuration:**
https://github.com/ffac/site/tree/v2023.2.2-1

**Custom patches:**
https://github.com/ffac/site/tree/v2023.2.2-1/patches

neocturne, 2025-03-30:
Please check if this still happens with current Gluon main, and provide more information if it does.

Djfe, 2025-11-18:
It's likely that this is solved in gluon main. The affected devices used v2023.2.x
Will reopen, if I ever encounter it again. Haven't encountered it either in the last months (used the device twice successfully since July 2025)

closed 2025-11-18 by Djfe