freifunk-gluon/gluon #3389: Fritz!Box 4040 eth0 mac is random on fresh installs

Djfe, 2024-12-13:
@herbetom reported the eth0 mac to be random since merging gluon next
https://paste.linuxlounge.net/QA

it might be related to this commit:
https://github.com/openwrt/openwrt/commit/b22d382ae4eaa1af42930115d91855f402314cac

maybe something broke the label mac since then or the label mac never worked at all

Djfe, 2024-12-13:
Update:
thanks to @neocturne
```
[13:26] <neocturne> djfe: the problem with the MAC address on ipq40xx is the missing alias ethernet0 in the Device Tree, which is used by the bootloader to fill in MAC addresses. It was previously incorrectly set in the SoC DTSI, but it belongs in in the board DTS(I). When it got removed from the SoC DTSI due to the update to Linux 6.6, nobody fixed it in the board DTS.
```

Djfe, 2024-12-13:
related
https://github.com/chunkeey/FritzBox-4040-UBOOT/pull/11

rotanid, 2024-12-30:
this might fix it:
https://github.com/openwrt/openwrt/pull/17432

maurerle, 2025-01-03:
I could not reproduce this issue on a FB7530, as my MAC-Address stays the same throughout boots and fresh installs.

Though I found a different MAC-Address related issue for my FB7530, mentioned in #3410

rotanid, 2025-01-08:
fixed in OpenWrt main, waiting for backport to 24.10 probably?
https://github.com/openwrt/openwrt/pull/17442

PR #3421 (2025-01-10): modules: update latest openwrt head
update openwrt to latest commit on openwrt-24.10

fixes #3389 
fixes #3410
closed 2025-01-10 by rotanid