freifunk-gluon/gluon #3391: main: hostapd reload fails

maurerle, 2024-12-14:
Using hostapd reload fails.
This is used in [ffac-ssid-changer](https://github.com/freifunk-gluon/community-packages/tree/master/ffac-ssid-changer)

```
ubus call hostapd reload
Command failed: Unknown error
```

in logread, we see:
```
Exception in ubus function: access to undeclared variable radio In [anonymous function](), file /usr/share/hostap/hostapd.uc, line 830, byte 45:
called from anonymous function (/usr/share/hostap/hostapd.uc:789:22)   `
    let config = iface_load_config(phy.phy, radio, phy.orig_file);` 
Near here ----------------------------------------------^
```

This happens in:
https://github.com/openwrt/openwrt/blob/aa1241e9794c2c03b74f14fef8530507ee4fc2f7/package/network/services/hostapd/files/hostapd.uc#L830

replacing this with `phy.radio_idx` on the device in `/usr/share/hostap/hostapd.uc` did not help

maurerle, 2024-12-14:
`phy.radio_idx`  fixes this - one just has to reboot for this

Neustradamus, 2024-12-19:
OpenWrt merged commit:
- https://github.com/openwrt/openwrt/commit/31e45f62cafbc9361c72d320241e589b31e2b5ed

closed 2024-12-23 by maurerle