freifunk-gluon/gluon #3393: main: additional client wifi interface requires wpa-supplicant-mini

maurerle, 2024-12-15:
### Bug report

**What is the problem?**

adding an additional wifi interface which connects as client on one radio to use WAN fails.

This is fixed by installing `wpa-supplicant-mini`.

Wifi Config:

```
config wifi-iface 'wan_radio0'
	option ifname 'wan0'
	option network 'fallback'
	option disabled '0'
	option device 'radio0'
	option mode 'sta'
	option encryption 'psk2'
    option ssid 'YOUR-WAN-SSID'
	option key 'your-wan-wifi-password'
	option macaddr '9a:96:1d:92:64:7b'
```

issuing wifi gives the following error:

```
Sun Dec 15 19:46:04 2024 daemon.notice hostapd: Set new config for phy phy0:
Sun Dec 15 19:46:04 2024 daemon.notice netifd: radio0 (6979): Command failed: Not found
Sun Dec 15 19:46:04 2024 daemon.notice netifd: Wireless device 'radio0' is now down
Sun Dec 15 19:46:04 2024 daemon.notice netifd: radio0 (6992): WARNING: Variable 'data' does not exist or is not an array/object
Sun Dec 15 19:46:05 2024 daemon.warn wgpeerselector: DNS resolution of ntp server a.ntp.ffac.rocks failed: error code -3
Sun Dec 15 19:46:05 2024 daemon.notice hostapd: Set new config for phy phy0:
Sun Dec 15 19:46:05 2024 daemon.notice netifd: radio0 (6992): Command failed: Not found
Sun Dec 15 19:46:05 2024 daemon.notice netifd: radio0 (6992): sh: out of range
Sun Dec 15 19:46:05 2024 daemon.notice hostapd: Set new config for phy phy0:
Sun Dec 15 19:46:05 2024 daemon.notice netifd: radio0 (6992): command failed: No such device (-19)
Sun Dec 15 19:46:05 2024 daemon.notice netifd: Wireless device 'radio0' is now up
```

I did not find any occurance of `wpa_supplicant` in `netifd` of openwrt, but I think it is called through hostapd or mac80211.sh ?

**What is the expected behaviour?**

Device should be available even without `wpa-supplicant-mini`

**Gluon Version:**

main with patches for hostapd reload

**Site Configuration:**

[ffac](https://github.com/ffac/site/)

**Custom patches:**

blocktrron, 2024-12-16:
STA interfaces providing uplink for VPN connections is not a supported operation mode of Gluon.

Hence closing this issue. If community-packaged rely on this behavior, ping me so i can move the issue over there.

closed 2024-12-16 by blocktrron

maurerle, 2024-12-16:
Sure, it is not needed for Gluon.
A fix has to be done in Openwrt anyway.
Is this by design or is it still possible to define STA interfaces without wpa-supplicant in openwrt-24.10?

I would like to at least improve the error message about missing wpa-supplicant in openwrt?

blocktrron, 2024-12-16:
As you expect a connection using WPA2, you surelay need either wpad or wpa-supplicant.