freifunk-gluon/gluon #3409: core: reconfigure gives warning: Failed to find NVMEM device

maurerle, 2025-01-01:
running gluon-reconfigure on main gives

```
Cannot parse config file '/etc/fw_env.config': No such file or directory
Failed to find NVMEM device
```

on my D-Link COVR-X1860 - this was not shown in v2023.2.x before - so it is a change coming from 24.10

I did not experience any issue, though this is more a cosmetic change.
This seems to come from the opkg package `uboot-envtools` which tries to configure a ramfs?

maurerle, 2025-01-01:
Dmesg log of a fresh boot:

```
[    0.000000] Linux version 6.6.67 (ffac@ffac-build) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r28179+6-aa1241e979) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Tue May  7 18:44:27 2024
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is D-Link COVR-X1860 A1
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] percpu: Embedded 12 pages/cpu s18688 r8192 d22272 u49152
[    0.000000] pcpu-alloc: s18688 r8192 d22272 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00010100
[    0.000000] Readback ErrCtl register=00010100
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 65024
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 248000K/262144K available (7803K kernel code, 628K rwdata, 1700K rodata, 1248K init, 218K bss, 14144K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.008087] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066227] pid_max: default: 32768 minimum: 301
[    0.077245] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.084494] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.101308] RCU Tasks Trace: Setting shift to 2 and lim to 1 rcu_task_cb_adjust=1 rcu_task_cpu_ids=4.
[    0.111146] rcu: Hierarchical SRCU implementation.
[    0.115853] rcu: 	Max phase no-delay instances is 1000.
[    0.121896] smp: Bringing up secondary CPUs ...
[    0.127277] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.127307] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.127321] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.127374] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.181386] Synchronize counters for CPU 1: done.
[    0.214022] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.214046] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.214060] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.214097] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.272985] Synchronize counters for CPU 2: done.
[    0.303736] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.303762] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.303775] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.303817] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.358205] Synchronize counters for CPU 3: done.
[    0.388094] smp: Brought up 1 node, 4 CPUs
[    0.399416] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.409232] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.416203] pinctrl core: initialized pinctrl subsystem
[    0.424812] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.431622] thermal_sys: Registered thermal governor 'step_wise'
[    0.433396] /pinctrl: Fixed dependency cycle(s) with /pinctrl/pinctrl0
[    0.446312] mt7621-pinctrl pinctrl: there is not valid maps for state default
[    0.457430] FPU Affinity set after 11720 emulations
[... 332 lines cut ...]
[   46.547397] batman_adv: bat0: No MLD Querier present - multicast optimizations disabled
[   46.656785] batman_adv: bat0: Adding interface: primary0
[   46.662636] batman_adv: bat0: Interface activated: primary0
[   46.736019] batman_adv: bat0: Interface deactivated: primary0
[   46.781072] batman_adv: bat0: Interface activated: primary0
[   46.977635] mt7530-mdio mdio-bus:1f internet: Link is Up - 1Gbps/Full - flow control rx/tx
[   46.977756] br-wan: port 1(internet) entered blocking state
[   46.991725] br-wan: port 1(internet) entered forwarding state
[   48.306674] batman_adv: bat0: Adding interface: mesh-vpn
[... cut ...]
[   50.779193] mt7915e 0000:02:00.0 client0: entered promiscuous mode
[   50.950429] br-client: port 4(client0) entered blocking state
[   50.956273] br-client: port 4(client0) entered forwarding state
```

Note the:
`unable to load configuration (fstab: Entry not found)` parts which was not there before

From my experience the fstab error were noticable on COVR-X1860, O2 6431, Raspberry Pi 3... So I think this is a general issue.

While the above NVMEM error was only noticable for me on COVR-X1860 - so is probably an upstream error..?

blocktrron, 2025-01-02:
The entry in the kernel ringbuffer is unrelated to the message displayed when reloading the board configuration.

First message is displayed due to invocation of  `fw_loadenv` in the `board_detect` stage and can be ignored.

The message about missing fstab can also be safely ignored, as OpenWrt uses it's own routine to mount the rootfs as well as overlayfs.

maurerle, 2025-01-03:
Thanks for clarifying!
Is there a simple way to remove the openwrt routine?
I think that the fstab part is quite confusing and appears on all installations.

Otherwise we can close this issue :)

closed 2025-01-10 by maurerle