freifunk-gluon/gluon #3410: fb7520/fb7530: MAC-Address does not match CWMP-Address

maurerle, 2025-01-03:
I just checked this on a FB7530, which I flashed fresh with Gluon - the MAC-Address for the FB7530 stays consistent throughout booting or flashing with -n, but does not match the last numbers of the "CWMP-Account" (which typically ends with the MAC-Address).

I did this to check for #3389 and the fix proposed in https://github.com/openwrt/openwrt/pull/17442

```
CWMP-Address: 0004E-B0F2083497A8
eth0 mac-address: b0:f2:08:34:97:a8
br-client mac-address: b0:f2:08:34:97:aa
```

The device appears on the map with `b0f2083497aa` (the wrong address)

`cat /proc/device-tree/aliases/ethernet0` -> `/soc/ethernet@c080000`
`hexdump -C /sys/firmware/devicetree/base/soc/ethernet@c080000/mac-address` shows the correct address (which exists on eth0 as well).

Not all FB7520/7530 have this CWMP-Address printed, and #2125 stated that the device does not have a label - so initial addition was done with a device without CWMP-Address.

So this requires the same fix as for the FB4040 in https://github.com/openwrt/openwrt/commit/b22d382ae4eaa1af42930115d91855f402314cac for the 7530 as well.

This behavior is not new, but happens with v2023.2.x as well, but is not severe, as most devices do not have a MAC-Address printed on them.

Djfe, 2025-01-03:
good find

rotanid, 2025-01-08:
@maurerle so should this be fixed in main due to merge of #3411 ?
is a backport to a gluon stable branch necessary?

maurerle, 2025-01-08:
Technically, the current state on gluon main is now the same as for the FB4040 (random mac)
The mac is bow correctly aken from eth0 which is random until the fixing commit is backported.
So this is as well waiting for a  backport of
https://github.com/openwrt/openwrt/pull/17442

A backport to a gluon stable branch is not needed from my point of view

PR #3421 (2025-01-10): modules: update latest openwrt head
update openwrt to latest commit on openwrt-24.10

fixes #3389 
fixes #3410
closed 2025-01-10 by rotanid