freifunk-gluon/gluon #3415: ath79: migrate GPIO numbering with OpenWrt 24.10

blocktrron, 2025-01-08:
OpenWrt 24.10 changed the starting index of GPIO lines of the SoC:

https://github.com/openwrt/openwrt/commit/ee01666cd32e156214fe1ebc250c5c9ed7256aba

It needs to be verified if we need to migrate those. One potential device is the CPE210 and its PoE passthrough switch.

maurerle, 2025-01-15:
I tested this on a CPE210.

Activating PoE passthrough did work fine.
The GPIO pin was successfully migrated to the new Pins

```
system.poe_passthrough=gpio_switch
system.poe_passthrough.name='PoE Passthrough'
system.poe_passthrough.gpio_pin='532'
system.poe_passthrough.value='1'
```

I did update from v2022 to gluon-main with poe disabled which kept the setting.
Did enable poe successfully.
Did downgrade to v2023.2.3 successfully with keeping the poe setting.
Did upgrade back to gluon main with keeping the PoE setting successfully.

So it looks like everything works fine with the GPIO change for the PoE passthrough

neocturne, 2025-03-11:
We appear to be missing code for migrating the `poe_passthrough` value from `system_gluon-old` to `system` at all, so while the GPIO number is updated, the value would always be the default from `site.conf` instead of the previous one.

maurerle, 2025-03-16:
@neocturne in the comment above, I mentioned that I enabled and disabled it correctly and the settings were kept.

During the gluon meeting we did not find the code where the setting is migrated.
This is due to `poe_passthrough` being the wrong keyword to search for, while the code is found for a `gpio_switch`:

https://github.com/freifunk-gluon/gluon/blob/91264654531969a32047993686de5e8854dfb24c/package/gluon-core/luasrc/lib/gluon/upgrade/002-migrate-system#L17-L27

This looks fine to me.
We can retest this, but I am now confident, that we are not falling back to the default on sysupgrades.

neocturne, 2025-03-23:
@maurerle Thanks, I think we can close this.

closed 2025-03-23 by neocturne

Djfe, 2025-04-09:
This new change in main to might be relevant here:
https://github.com/openwrt/openwrt/commit/0ac79009dd3433d5e92a8311beae1041f39c3b79

Can someone please take a look whether this change might be required for 24.10 as well to avoid issues?