freifunk-gluon/gluon #3423: [armsr] device name might be renamed upstream

Djfe, 2025-01-15:
Just a note to keep track of the following upstream PR
https://github.com/openwrt/openwrt/pull/12963

We might have to adjust the device name to `combined-efi` and add a manifest alias if it gets merged upstream (and possibly backported)
https://github.com/freifunk-gluon/gluon/blob/main/targets/armsr.inc

Djfe, 2025-01-20:
If I find the time, I'll backport blocktrron's commit to v2023.2.x
https://github.com/freifunk-gluon/gluon/commit/8f8536681f06ec845df1b6abe358594bbb63d3e0 

Do we require a manifest alias for the fix to be complete?

blocktrron, 2025-01-20:
We don't need to do anything, as the OpenWrt image-name exclusive change does not affect the image name the autoupdater expects.

closed 2025-01-20 by blocktrron