freifunk-gluon/gluon #3425: [ath79][ipq40xx] verify 5GHz on devices using ath10k QCA9888

Djfe, 2025-01-15:
This a crossreference to an upstream ticket.
The nvmem-cell rework caused certain devices to have missing 5GHz radios. Specifically devices with a chip requiring QCA9888 firmware (wifi5 wave2).
https://github.com/openwrt/openwrt/issues/14541

Affected are likely (untested so far):
TP-Link
- EAP225 variants (outdoor for example)
- Archer C58 v1, C59 v1, C6 v2

ZTE MF281

OpenMesh A62

Plasma Cloud PA2200

(and Linksys Velop WHW03 v1/v2, which we were planning on adding, though I haven't noticed the issue on these while we were testing)

matjon, 2025-01-21:
Hello,

I'm unable to test my Tp-Link Archer C59 v1 right now, but on **OpenWRT 23.05.5** in `dmesg` I get:

```
Sat Dec  7 10:14:26 2024 kern.info kernel: [   16.345023] ath10k 5.15 driver, optimized for CT firmware, probing pci device: 0x56.
Sat Dec  7 10:14:26 2024 kern.info kernel: [   16.354794] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
Sat Dec  7 10:14:26 2024 kern.info kernel: [   16.361624] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
Sat Dec  7 10:14:26 2024 kern.info kernel: [   18.895374] ath10k_pci 0000:00:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
Sat Dec  7 10:14:26 2024 kern.info kernel: [   18.904962] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
Sat Dec  7 10:14:26 2024 kern.info kernel: [   18.924148] ath10k_pci 0000:00:00.0: firmware ver 10.4b-ct-9888-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 59e741e7
Sat Dec  7 10:14:26 2024 kern.info kernel: [   19.257555] ath10k_pci 0000:00:00.0: Loading BDF type 0
Sat Dec  7 10:14:26 2024 kern.err kernel: [   19.268512] ath10k_pci 0000:00:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=20 from ath10k/QCA9888/hw2.0/board-2.bin
Sat Dec  7 10:14:26 2024 kern.info kernel: [   19.486912] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id 0:20 crc32 3f3d8ea0
Sat Dec  7 10:14:26 2024 kern.warn kernel: [   21.342408] ath10k_pci 0000:00:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Sat Dec  7 10:14:26 2024 kern.warn kernel: [   21.350217] ath10k_pci 0000:00:00.0: msdu-desc: 2500  skid: 32
Sat Dec  7 10:14:26 2024 kern.info kernel: [   21.380347] ath10k_pci 0000:00:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Sat Dec  7 10:14:26 2024 kern.info kernel: [   21.391723] ath10k_pci 0000:00:00.0: wmi print 'free: 114572 iram: 12644 sram: 29508'
Sat Dec  7 10:14:26 2024 kern.info kernel: [   21.629032] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
```

matjon, 2025-01-22:
I have tested and the firmware file is indeed missing:

```
Wed Jan 22 21:37:50 2025 kern.info kernel: [   19.700473] ath10k 6.10 driver, optimized for CT firmware, probing pci device: 0x56.
Wed Jan 22 21:37:50 2025 kern.info kernel: [   19.764397] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
Wed Jan 22 21:37:50 2025 kern.info kernel: [   19.771176] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
Wed Jan 22 21:37:50 2025 kern.info kernel: [   23.266864] ath10k_pci 0000:00:00.0: qca9888 hw2.0 target 0x01000000 chip_id 0x00000000 sub 0000:0000
Wed Jan 22 21:37:50 2025 kern.info kernel: [   23.276444] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
Wed Jan 22 21:37:50 2025 kern.info kernel: [   23.290922] ath10k_pci 0000:00:00.0: firmware ver 10.4b-ct-9888-fW-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 59e741e7
Wed Jan 22 21:37:50 2025 kern.err kernel: [   23.668764] ath10k_pci 0000:00:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=20 from ath10k/QCA9888/hw2.0/board-2.bin
Wed Jan 22 21:37:50 2025 kern.err kernel: [   24.184399] ath10k_pci 0000:00:00.0: failed to fetch board-2.bin or board.bin from ath10k/QCA9888/hw2.0
Wed Jan 22 21:37:50 2025 kern.err kernel: [   24.194183] ath10k_pci 0000:00:00.0: failed to fetch board file: -12
Wed Jan 22 21:37:50 2025 kern.err kernel: [   24.223641] ath10k_pci 0000:00:00.0: could not probe fw (-12)
```

I have fixed it with `boarddata_0.bin` from the vendor's image:

```
./ath10k-bdencoder --add-board board-2.bin squashfs-root/lib/firmware/QCA9888/hw.1/boarddata_0.bin 'bus=pci,bmi-chip-id=0,bmi-board-id=20'
```

Djfe, 2025-01-26:
Maybe I should've specified, that this issue is only happening on OpenWrt 24.10 (or gluon main branch)

rotanid, 2025-06-10:
for one of the devices, Archer C6 v2, it has been solved via https://github.com/openwrt/openwrt/pull/19035
according to the issue comment https://github.com/openwrt/openwrt/issues/14541#issuecomment-2948520338 , it's "only" a matter of someone working on other devices, not magic ?

achterin, 2025-10-05:
looks like tp-link eap225-outdoor devices are unaffected. we've tested at least version 3 and it was able to load the boarddata just fine.

achterin, 2025-10-06:
> [@achterin](https://github.com/achterin) you mentioned on [#3587](https://github.com/freifunk-gluon/gluon/pull/3587) that c60 v2 was also affected?
> 
> maybe we need an updated list on which device is fixed and which is still affected?

yeah, an updated list would be desirable to track the process. at least for the devices included in gluon.
i can't say with 100% certainty that other devices are affected, because i dont own any of them. i just went through the openwrt bugreports and tried to provide testing images for people.

these are the devices with existing bugreports:

- TP-Link Archer C59 v1
- TP-Link Archer C60 v2 (not included in gluon)
- TP-Link Archer C60 v3 (not included in gluon)

most likely the Archer C58 v1 will have the same issue. as @moridius was pinged in the pr adding support (back) for this device, maybe he can help here as well.

as already said: the v3 of the EAP225-Outdoor worked without any issues on the last OpenWrt snapshot, so most likely the other version are unaffected as well.

achterin, 2025-10-18:
funny thing is: its already marked as broken due to 64MB which will oom in almost every scenario. on the other hand: if the second radio is not detected anymore, 64MB should be enough for single-band use xD
in all seriousness, i agree with you, add a remark and be done for now.

closed 2025-10-22 by rotanid