freifunk-gluon/gluon #3439: GL.iNet GL-MT6000 (Flint2) - Gluon hardware support?

phqzgunsfjror, 2025-02-06:
Hi!

I found an OpenWrt [table](https://openwrt.org/toh/views/toh_gluon_supported) showing Gluon supported hardware. It lists the GL.iNet GL-MT6000 (Flint2) but it is not listed [here](https://gluon.readthedocs.io/en/latest/user/supported_devices.html).

I just want to ask if this router is supported or not...

blocktrron, 2025-02-06:
The router is currently not supported, all supported devices are listed in the documentation.

If the device is supported by OpenWrt, you can read on the requirements for Gluon support here: https://gluon.readthedocs.io/en/latest/dev/hardware.html

closed 2025-02-06 by blocktrron

rotanid, 2025-02-10:
someone called "raylynnknight" addded the wrong information on 2024-03-26, i corrected it now.