freifunk-gluon/gluon #3450: Revisit shellcheck excludes

grische, 2025-03-10:
[Currently, the Gluon CI excludes](https://github.com/freifunk-gluon/gluon/blob/e2de86ce9dfaaf3f5aac890bd1a0cedbacf25dc2/scripts/lint-sh.sh#L20)

- [SC2039](https://www.shellcheck.net/wiki/SC2039): `This warning has been retired in favor of individual SC3xxx warnings for each individual issue. Removed in V0.7.2 - 2021-04-19`
- [SC1091](https://www.shellcheck.net/wiki/SC1091): Needs a revisit
- [SC2155](https://www.shellcheck.net/wiki/SC2155): Needs a revisit
- [SC2034](https://www.shellcheck.net/wiki/SC2034): Probably required for scripts that are embedded within other scripts. Maybe this can be handled differently to find undefined variables?
- [SC3043](https://www.shellcheck.net/wiki/SC3043): Can be deprecated with shellcheck v0.10.0 with the busybox sh interpeter
- [SC3037](https://www.shellcheck.net/wiki/SC3037): Can be deprecated with shellcheck v0.10.0 with the busybox sh interpeter
- [SC3057](https://www.shellcheck.net/wiki/SC3057): Can be deprecated with shellcheck v0.10.0 with the busybox sh interpeter

cc @AiyionPrime

AiyionPrime, 2025-03-10:
thanks, will look into it in the next days.

neocturne, 2025-03-11:
I think keeping SC2039 for now makes sense if Gluon is still buildable on Debian oldstable or Ubuntu 20.04, as these distros ship shellcheck 0.7.1 and 0.7.

SC1091, SC2155 and SC2034 should be dropped from the excludes.

neocturne, 2025-03-19:
I have also started looking into fixing the scripts, and I actually found some weirdness in the wireguard protocol handler after enabling the lints. I'll open a PR soon.

closed 2025-03-31 by rotanid