freifunk-gluon/gluon #3459: Verify D-Link DGS-1210-10P device name

neocturne, 2025-03-24:
Is the device name `d-link_dgs-1210-10p` correct (matches `lua -e 'print(require("platform_info").get_image_name())'`)? For most targets it should be impossible for an underscore to appear because of the way the name is derived from the model name.

_Originally posted by @neocturne in https://github.com/freifunk-gluon/gluon/pull/2707#discussion_r2010814717_

neocturne, 2025-03-24:
@AiyionPrime

AiyionPrime, 2025-03-24:
Can't say for sure, I do not own the device anymore.
Had it for a customer.
I can check it in two months earliest.

neocturne, 2025-03-25:
@AiyionPrime Are the any devices of the model online?

AiyionPrime, 2025-03-25:
@neocturne Currently not, no.
I liked the idea of a switch being capable of batman-adv.
As the introduction of port isolation would result in more accurate topology depictions in e.g. the meshviewer.

As a matter of fact this device is ill suited for the purpose due to its terrible CPU.

I thought I'd removed this a long time ago from gluon, but will do later this evening if there are no objections.

neocturne, 2025-03-25:
Fair enough, removing the device also solves the issue ^^

AiyionPrime, 2025-03-25:
If I ever need it again, I'll re-add it with the proper name :)

PR #3461 (2025-03-28): hardware: remove support for D-Link DGS-1210-10P and realtek-rtl838x target
This device should never have been supported as non-broken. It is not a useful target for batman-adv (as the CPU is too slow for reasonable throughput; it may be more useful with L3 routing protocols); also, the autoupdater never worked because of an incorrect image name.

As there it nobody actually running this device (or planning to do so at the moment), just drop the device instead of fixing the name. The realtek-rtl838x target becomes empty and is dropped.

Fixes #3459
closed 2025-03-28 by neocturne