freifunk-gluon/gluon #3466: OLSR: incorrect handling of wired mesh

neocturne, 2025-03-30:
Looking at `package/gluon-mesh-olsrd/luasrc/lib/gluon/upgrade/360-gluon-mesh-olsrd-setup-intf`, the script duplicates a lot of logic from our generic interface role handling - which is a problem, as it limits us in extending the generic role handling code. In addition, there seems to be a very concrete issue with the code: it directly refers to the `mesh_uplink` and `mesh_other` interfaces, which is incorrect for VXLAN-enabled wired mesh.

For possible solutions, I don't know enough about olsrd2 - is it possible to change the interface list at runtime, or is it necessary restart the daemon?

**Gluon Version:** main (v2023.2-374-g715abd7f)

cc @mkg20001

Djfe, 2025-11-19:
Is this a blocker for the next release actually? Probably not, I assume.
But it will become relevant again afterwards if we switch to VRF, right?

Djfe, 2025-11-19:
related
https://forum.freifunk.net/t/olsrv2-erfahrungen/22272/2

and I put the word out there, even though I assume that Funkfeuer is currently the only user of OLSR:
https://forum.freifunk.net/t/olsr-support-in-gluon/24516

mkg20001, 2026-09-02:
olsrv1 has a patch for ubus supoort for dynamically adding interfaces. olsrdv2 was removed from openwrt, it was also really buggy and unmantained.

How does this work in batman? How are interfaces added dynamically?

mkg20001, 2026-09-02:
I will do a PR to replace olsr2 with olsrd/olsrd6 upstream, to keep olsr working, and as part of that I will refactor this to work properly with the way it should in gluon.