freifunk-gluon/gluon #3481: mt76: mt7915: 5 GHz radio becomes deaf

blocktrron, 2025-04-14:
### Bug report

Take this as a "tracking Issue" for myself

**What is the problem?**

The 5GHz radio on a MT7915 device becomes "deaf". It reports single-digit airtime utilization in a busy environment, no clients and a monitor interface shows zero packets.

The device (however) still scans and detects networks.

Firmware debug is spammed with

```
```

No packets on monitor interface

```
root@64283-cccda-w:/sys/kernel/debug/ieee80211/phy1/mt76# date; tcpdump -i monitor1
Mon Apr 14 04:21:10 CEST 2025
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on monitor1, link-type IEEE802_11_RADIO (802.11 plus radiotap header), snapshot length 262144 bytes
^C
0 packets captured
0 packets received by filter
0 packets dropped by kernel
root@64283-cccda-w:/sys/kernel/debug/ieee80211/phy1/mt76# date
Mon Apr 14 04:21:13 CEST 2025
root@64283-cccda-w:/sys/kernel/debug/ieee80211/phy1/mt76# 
```

Interface config

```
root@64283-cccda-w:/sys/kernel/debug/ieee80211/phy1/mt76# iw dev
phy#1
	Interface monitor1
		ifindex 24
		wdev 0x100000004
		addr b8:ec:a3:dd:a5:d5
		type monitor
		txpower 26.00 dBm
	Interface owe1
		ifindex 23
		wdev 0x100000003
		addr c6:2f:3b:12:9b:be
		ssid owe.darmstadt.freifunk.net
		type AP
		channel 100 (5500 MHz), width: 20 MHz, center1: 5500 MHz
		txpower 26.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
	Interface client1
		ifindex 18
		wdev 0x100000001
		addr c6:2f:3b:12:9b:bc
		ssid darmstadt.freifunk.net
		type AP
		channel 100 (5500 MHz), width: 20 MHz, center1: 5500 MHz
		txpower 26.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	7673312	60185	0	373	6721514	531698051		8100330
phy#0
	Interface monitor0
		ifindex 25
		wdev 0x4
		addr b8:ec:a3:dd:a5:d4
		type monitor
[... 672 lines cut ...]
[4746247.696978] ieee80211 phy0: WM: (2788.004297:92:MURU-E)[muruSwPpduDurAdapt] score = 0 Fail 
[4746247.697018] ieee80211 phy0: WM: (2788.004420:93:MURU-E)[muruSwPpduDurAdapt] score = 0 Fail 
[4746247.697048] ieee80211 phy0: WM: (2788.004450:94:MURU-E)[muruSwPpduDurAdapt] score = 0 Fail 
[4746247.697076] ieee80211 phy0: WM: (2788.004450:95:MURU-E)[muruSwPpduDurAdapt] score = 0 Fail 
[4746247.697106] ieee80211 phy0: WM: (2788.004481:96:MURU-E)[muruSwPpduDurAdapt] score = 0 Fail 
[4746247.698670] ieee80211 phy0: WM: (2788.006037:97:MURU-E)[muruSwPpduDurAdapt] score = 0 Fail 
[4746247.698694] ieee80211 phy0: WM: (2788.006067:98:MURU-E)[muruSwPpduDurAdapt] score = 0 Fail 
[4746247.698709] ieee80211 phy0: WM: (2788.006098:99:MURU-E)[muruSwPpduDurAdapt] score = 0 Fail 
[4746247.698726] ieee80211 phy0: WM: (2788.006128:00:MURU-E)[muruSwPpduDurAdapt] score = 0 Fail 

```

**What is the expected behaviour?**

Radio should be working

**Gluon Version:**

Gluon master (24.10 based)

**Site Configuration:**

https://github.com/freifunk-darmstadt/site-ffda/commit/fc9c386

**Custom patches:**

Djfe, 2025-04-15:
You might want to edit your first post and add, what "Firmware debug is spammed with" is exactly. currently it's empty.
I assume `score = 0 Fail`
I'll hide this as resolved once you saw my comment

maurerle, 2025-09-21:
I still experienced this issue today with an NWA55AXE I now have to test.
Tested with latest gluon-main.

Looking at `iw phy phy1 get txq` shows that there are packets in the backlog for this radio.
On the first sight, this looks related to this issue.

blocktrron, 2025-09-21:
Please check output of 

```
/sys/kernel/debug/ieee80211/phy0/mt76/xmit-queues
/sys/kernel/debug/ieee80211/phy0/mt76/rx-queues
```

maurerle, 2025-09-22:
So to describe the error behavior more precise:
1. ping to link-local address has times of 100ms to multiple seconds
2. `iw phy phy1 get txq` shows a backlog of greater than 0
3. commands you provided:

```
root@ffac-lothringer-nwa55axe:~# cat /sys/kernel/debug/ieee80211/phy0/mt76/xmit-queues
     queue | hw-queued |      head |      tail |
      MAIN |         1 |       114 |       113 |
     MCUWM |         0 |        19 |        19 |
     MCUWA |         0 |       152 |       152 |
   MCUFWDL |         0 |       119 |       119 |
root@ffac-lothringer-nwa55axe:~# cat /sys/kernel/debug/ieee80211/phy0/mt76/rx-queues
     queue | hw-queued |      head |      tail |
         0 |      1535 |      1033 |      1034 |
         1 |       511 |        18 |        19 |
         2 |       511 |       307 |       308 |
         3 |      1535 |      1510 |      1511 |
         4 |       511 |       511 |         0 |
root@ffac-lothringer-nwa55axe:~# iw phy phy1 get txq | grep Backlog
Backlog:		260 pkts
```
4. after `echo 1 > /sys/kernel/debug/ieee80211/phy1/mt76//fw_debug_wm`  the log is spammed with `muruSwPpduDurAdapt] score = 0 Fail`
5. besides of this, I see the following logs:
```
Mon Sep 22 19:12:44 2025 kern.info kernel: [29624.473781] ieee80211 phy0: WA: StaRec fgIsInUse is already FALSE. Do nothing
Mon Sep 22 19:12:44 2025 daemon.info hostapd: client1: STA d6:73:89:63:d0:90 IEEE 802.11: authenticated
Mon Sep 22 19:12:44 2025 daemon.notice hostapd: client1: STA-OPMODE-N_SS-CHANGED d6:73:89:63:d0:90 2
Mon Sep 22 19:12:44 2025 daemon.info hostapd: client1: STA d6:73:89:63:d0:90 IEEE 802.11: associated (aid 2)
Mon Sep 22 19:12:44 2025 kern.info kernel: [29624.478470] ieee80211 phy0: WM: (3824.618161:04:BSS-E)_whCapSetGeneric_Falcon DW0 = 0x0, DW1 = 0x0
Mon Sep 22 19:12:44 2025 kern.info kernel: [29624.479011] ieee80211 phy0: WM: (3824.618679:05:TXC-E)heACtrlInitStaRec(): prOperMode=401b58, u2OperMode=41!
Mon Sep 22 19:12:44 2025 kern.info kernel: [29624.479061] ieee80211 phy0: WM: (3824.618771:06:BSS-E)muruAddStaRec, prRuStaRec->u1Bw = 0 eBand = 5000000 BSS BW = 0
Mon Sep 22 19:12:44 2025 daemon.notice hostapd: client1: AP-STA-CONNECTED d6:73:89:63:d0:90 auth_alg=open
Mon Sep 22 19:12:44 2025 daemon.notice hostapd: client0: Prune association for d6:73:89:63:d0:90
Mon Sep 22 19:12:44 2025 daemon.notice hostapd: client0: AP-STA-DISCONNECTED d6:73:89:63:d0:90
Mon Sep 22 19:12:44 2025 kern.info kernel: [29624.485861] ieee80211 phy0: WA: ==>rxmDeleteStaRec, prStaRecord->fgIsInUse = 1
Mon Sep 22 19:12:44 2025 kern.info kernel: [29624.486857] ieee80211 phy0: WM: (3824.626522:07:BSS-E)_whCapSetGeneric_Falcon DW0 = 0x0, DW1 = 0x0
Mon Sep 22 19:12:44 2025 kern.info kernel: [29624.488873] ieee80211 phy0: WA: Invalid baDirection
Mon Sep 22 19:12:44 2025 daemon.info hostapd: client1: STA d6:73:89:63:d0:90 RADIUS: starting accounting session CA17886E1D3F64FE
Mon Sep 22 19:12:44 2025 kern.info kernel: [29624.623257] ieee80211 phy0: WA: Invalid baDirection
Mon Sep 22 19:12:44 2025 user.info usteer: station d6:73:89:63:d0:90 disconnected from node hostapd.client0
Mon Sep 22 19:12:44 2025 daemon.info hostapd: client1: STA c6:7a:f2:88:8c:fd IEEE 802.11: authenticated
Mon Sep 22 19:12:44 2025 kern.info kernel: [29625.059261] ieee80211 phy0: WM: (3825.198788:08:MURU-E)[muruMumDeauthCheck] u2WlanIdx=23,u1ApNsts=2, u1StaNsts=2
Mon Sep 22 19:12:44 2025 kern.info kernel: [29625.059584] ieee80211 phy0: WM: (3825.199154:09:BSS-E)_whCapSetGeneric_Falcon DW0 = 0x0, DW1 = 0x0
Mon Sep 22 19:12:44 2025 kern.info kernel: [29625.059656] ieee80211 phy0: WA: StaRec fgIsInUse is already FALSE. Do nothing
Mon Sep 22 19:12:44 2025 daemon.info hostapd: client1: STA c6:7a:f2:88:8c:fd IEEE 802.11: associated (aid 5)
Mon Sep 22 19:12:44 2025 daemon.notice hostapd: client1: AP-STA-CONNECTED c6:7a:f2:88:8c:fd auth_alg=open
Mon Sep 22 19:12:44 2025 kern.info kernel: [29625.068402] ieee80211 phy0: WM: (3825.207974:10:BSS-E)_whCapSetGeneric_Falcon DW0 = 0x0, DW1 = 0x0
Mon Sep 22 19:12:44 2025 kern.info kernel: [29625.069365] ieee80211 phy0: WM: (3825.208920:11:TXC-E)heACtrlInitStaRec(): prOperMode=401f80, u2OperMode=0!
Mon Sep 22 19:12:44 2025 daemon.info hostapd: client1: STA c6:7a:f2:88:8c:fd RADIUS: starting accounting session F6F4E8A348722364
Mon Sep 22 19:12:44 2025 kern.info kernel: [29625.076354] ieee80211 phy0: WM: (3825.213742:12:BSS-E)_whCapSetGeneric_Falcon DW0 = 0x0, DW1 = 0x0
Mon Sep 22 19:12:44 2025 kern.info kernel: [29625.087971] ieee80211 phy0: WA: Invalid baDirection
```
6. I do see a low mesh quality and tq, but noise and channel utilization reports valid values

maurerle, 2025-09-30:
I think my issue is something different than you described in you initial post.
I created an issue in mt76 describing my issue: https://github.com/openwrt/mt76/issues/1009

I must say that I never experienced the 5GHz radio becoming completely deaf on mt7915 and I really would have expected to run into this issue with the number of mt7915 devices we have in FFAC