freifunk-gluon/gluon #3483: ramips-mt7621: ethernet link unstable - race condition in ethernet driver triggered by EEE

Djfe, 2025-04-15:
### Bug report

**What is the problem?**
ramips-mt7621 currently has unstable ethernet for devices using the switch mt7530, where links flap on/off in some setups or are broken entirely unless using failsafe mode in OpenWrt. This is somehow related to energy efficient ethernet (EEE). The upstream maintainer for mt7530 sees the issue rooted elsewhere in the mediatek ethernet driver.
Upstream issue and possible workaround patch (only until an actual solution is found)
https://github.com/openwrt/openwrt/issues/17351#issuecomment-2800040994

some related posts on the mailinglist by said maintainer (just for reference)
https://lists.openwrt.org/pipermail/openwrt-devel/2025-April/043924.html
https://lists.openwrt.org/pipermail/openwrt-devel/2025-January/043671.html
https://lists.openwrt.org/pipermail/openwrt-devel/2025-January/043673.html
https://github.com/openwrt/openwrt/issues/17351#issuecomment-2609336415

This should be tracked on our [milestone](https://github.com/freifunk-gluon/gluon/milestone/24)
If we want to release early we might have to bite into a sour apple and disable EEE for this target. But we might just wait it out instead or help in contributing a fix.

**What is the expected behaviour?**
Ethernet should run stable on ramips-mt7621 to prevent nodes from going offline or even softbricking them (Zyxel NWA55AXE) after running an update to main or upcoming v2025.1.

**Gluon Version:**
main

**Site Configuration:**
[irrelevant](https://github.com/ffac/site/tree/v2025.1.x)

**Custom patches:**
[irrelevant](https://github.com/ffac/site/tree/v2025.1.x/patches)

AiyionPrime, 2025-06-20:
@blocktrron stellt in den Raum, dass das Upstream durch Deaktivierung des Features auf Empfehlung des Entwicklers gefixt wurde.

Eventuell ist das nur noch ein module update entfernt, ist er aber gerade nicht sicher.

Djfe, 2025-06-21:
yes, this is just one module update away :)
https://github.com/openwrt/openwrt/commit/926314ab12e4858477b35d152157339e9038dee6

rotanid, 2025-06-24:
@Djfe please test this, may be fixed now

PR #3529 (2025-06-24): modules: update to latest HEAD
this resolves #3483, as it contains 926314ab12,
which marks EEE as broken, and should result in EEE avoidance on the affected devices.

Testing is required, as this is linked to an issue.

@Djfe do you still have the device at hand and could build an image with this?