freifunk-gluon/gluon #3487: new build features SMALL_FLASH and LOW_MEMORY_FOOTPRINT

Djfe, 2025-04-17:
### Bug report

We should figure out how `SMALL_FLASH` and `LOW_MEMORY_FOOTPRINT` affect us.
The first drops atleast support for ED25519 on low flash devices but maybe there is more.
https://github.com/openwrt/openwrt/blob/openwrt-24.10/package/network/services/dropbear/Config.in#L52

Since we planned on stopping dropbear from creating new rsa host keys this is quite relevant to new installations or devices that are being reset by `firstboot`

side note about a new build option for dropbear:
Only allowing for modern (client) keys could be a nice option to set via site.json for new communities (opt-in)
https://github.com/openwrt/openwrt/commit/bf900e02c7102601be2e9280231711e70f065877

**What is the expected behaviour?**
Check that no other packets that affect gluon are changed by `SMALL_FLASH` or `LOW_MEMORY_FOOTPRINT`

Stuff I noticed (but there is certainly more):
some subtargets are marked as `SMALL_FLASH` like ramips-mt76x8 or ath79-tiny (this one we don't support) or xrx200_legacy (see https://github.com/freifunk-gluon/gluon/blob/main/targets/lantiq-xrx200_legacy , two devices currently) 

also see
https://github.com/openwrt/openwrt/blob/openwrt-24.10/config/Config-kernel.in#L1401

https://github.com/openwrt/openwrt/blob/openwrt-24.10/config/Config-images.in#L157

ujail is dropped for `SMALL_FLASH`, but we just started using ujail for dnsmasq in Gluon main
https://github.com/openwrt/openwrt/blob/openwrt-24.10/include/target.mk#L95

**Gluon Version:**
main

Djfe, 2025-04-17:
I dislike how they marked the full mt76x8 target as small

The target has many devices with 16 and 32MiB of flash after all
https://github.com/openwrt/openwrt/commit/fd46680966e9b9bc282b665282e2e6a22a27c26e

Djfe, 2025-04-17:
Neocturne took a look. currently a none issue for us so therefore closed

blocktrron, 2025-04-18:
@Djfe  please add more context please.

Reading the conversation form IRC, it is - in fact - no change with the current release in terms of the flag itself to dnsmasq, as this is from 2020 and contained in releases 21.02 onwards: https://github.com/openwrt/openwrt/commit/d0f295837a03f7f52000ae6d395827bdde7996a4

However the addition to mt76x8 (at least) is from this year and currently only in Gluon master:
https://github.com/openwrt/openwrt/commit/fd46680966e9b9bc282b665282e2e6a22a27c26e

Thus, i am 100% certain that there is in fact a regression. At least for dropbear, i would triage this as somewhat-critical as it results in users not being able to connect to devices they were previously able to connect to.

blocktrron, 2025-04-18:
Reopen until clarified.

neocturne, 2025-04-18:
@blocktrron You are right, this is not entirely irrelevant. I intend to take care of the dropbear configuration when moving to Ed25519-only host keys as discussed in the meetup.

blocktrron, 2025-04-18:
@neocturne understood. do you have a timeframe? no intention on putting up pressure, i just have a fix already staged for it locally, so if you plan to do it in the coming weeks, I'd suggest to put this on main in order to have it fixed for now.

neocturne, 2025-04-18:
@blocktrron Ah, if you just want to enable the ED25519 feature unconditionally, feel free to open a PR, that will have to be done anyways.

neocturne, 2025-04-18:
Regarding ujail - do we also want to include that unconditionally?

blocktrron, 2025-04-18:
@neocturne  I'd suggest to take this into focus once we've assessed if we can make use of it in the service gluon introduced by itself. I see respondd as a prime example.