freifunk-gluon/gluon #3489: status page integer overflow - traffic statistic

Djfe, 2025-04-20:
### Bug report

**What is the problem?**
status page integer overflow when accumulated traffic exceeds 2TiB (I assume due to 32bit signed integer)
starts counting at zero again

this has low priority though

**Gluon Version:**
v2023.2.4

**Site Configuration:**
https://github.com/ffac/site/tree/v2023.2.x

**Custom patches:**
https://github.com/ffac/site/tree/v2023.2.x/patches

neocturne, 2025-04-20:
@Djfe Please provide information on what hardware you're seeing this behavior.

Djfe, 2025-04-20:
on this Nanopi R4S
https://map.aachen.freifunk.net/#!/de/map/c2b9a2a1c0d2

neocturne, 2025-04-20:
Hmm, 2TiB is a very weird value for this to fail - all counters are in bytes, so even if there is a 32bit computation somewhere, it should be at 2 or 4 GiB. In addition, wrapping from 2 GiB to 0 is also unusual. Are you sure the network was running the whole time the node was up?

Djfe, 2025-04-21:
vpn was indeed disconnected yesterday around 2:00 in the morning due to me restarting interfaces on the supernode. could this have reset the traffic counters?
the nanopi probably connected to a different gateway when it happened.

neocturne, 2025-04-21:
@Djfe At a glance, [tecff-general-workaround](https://github.com/ffac/gluon-packages/tree/main/tecff-general-workaround) might have precisely that effect - there is a code path that calls `/etc/init.d/network restart`, and that will recreate bat0.

Djfe, 2025-04-21:
OK I didn't know how the counter was kept. It's attached to the network interface? I see that fully explains it, I think. Thanks
I should've correlated this when the owner asked me about it but it didn't occur to me

closed 2025-04-21 by Djfe

neocturne, 2025-04-21:
The traffic counters maintained internally by batman-adv and queried from the kernel using the ethtool ioctls. Like all network counters, they are part of interface state.