freifunk-gluon/gluon #3490: gluon-reconfigure interface roles: bad argument #1 to 'pairs' (table expected, got string)

Djfe, 2025-04-20:
### Bug report

**What is the problem?**
This is low priority, I'm going to write it down, so I don't forget about it.

This is about an error caused by the user setting interface-roles in `uci show gluon` with an unexpected type (string instead of table).
by using `uci set` instead of `uci add_list/del_list`

**What is the expected behaviour?**
We could add a proper error message here instead of confusing users, that tells the user how to set these correctly and how to fix the error.
Also if it's possible, we might want to parse the string and correct the error by translating it into a table instead of erroring at all. (atleast for the cases that are defined: uplink, mesh or client separated by a space or commas)
We could still show a warning at the end of running gluon-reconfigure to educate the user and/or tell them that the new configuration might not be what they intended to do.

Affected code is atleast in 021-interface-roles (line 59) and 110-network (line 33)
Both call simple-uci.lua:22 in the end.
110-network takes a detour by calling util.lua:123 first, which in return calls simple-uci.lua

**Gluon Version:**
v2023.2.4

**Site Configuration:**
[Neanderfunk](https://github.com/eulenfunk/firmware/tree/v2023.2.x)

**Custom patches:**
https://github.com/eulenfunk/firmware/tree/v2023.2.x/patches

blocktrron, 2025-04-20:
I don't see this as an issue, this is user-error in my opinion.

neocturne, 2025-04-20:
Yea, UCI doesn't have a schema we could validate against - if you set something invalid, that will cause issues.

blocktrron, 2025-04-20:
Closing, as this is not a bug.

closed 2025-04-20 by blocktrron