freifunk-gluon/gluon #3500: IPv6 router solicitations aren't received via Ethernet

herbetom, 2025-05-06:
### Bug report

**What is the problem?**

I can't yet point the finger anywhere concrete. I encounter the problem that IPv6 Router Solicitations don't seem to reach the Node (don't see them via tcpdump) if send via client ethernet Ports and therefore aren't answered by uradvd. Don't know yet if the situation is different for VPN vs no-vpn. Currently testing without vpn.

I receive Router Advertisments if a Client sends Router Solicitations via wifi. But those prefixes don't stay configured for more then a couple of seconds (maybe 10s).

Not sure yet wheter this is new issue or specific to devices. I think i've had the problem before but never dug deeper. But idk.

Currently testing with the following device:

```
Hostname:                       64625-3ccd5772ff17-Xiaomi-4a-GE
MAC address:                    3c:cd:57:72:ff:17
Hardware model:                 Xiaomi Mi Router 4A Gigabit Edition
Gluon version / Site version:   v2023.2-142-gca9dcd5 / 3.0.6
Firmware release:               3.0.6~20250505
Site:                           Freifunk Darmstadt
Domain:                         ffda_64673
Public VPN key:                 724259408885162fb5843095d0c99c9287700e3f21ed5c2eb529e0d718bfeaf8
Switch type:                    dsa
```

When Router Solicitations are sent from one PC i can see them in an tcpdump which is connected to the other Port of the Router. So they're send and the switch seems to work. But i don't seem them on the node itself on the br-client bridge.

Tools & Commands:

`rdisc6 enp42s0f1u1` to send Router Solicitations
`tcpdump -ttt icmp6 and \('ip6[40] = 134' or 'ip6[40] = 133'\) -vvvv -ni enp42s0f1u1` to see Router Solicitations and Advertisments-

**What is the expected behaviour?**

Router solicitations beeing received and answered with Router Advertisments

**Gluon Version:**

v22023.2.x - basically https://github.com/freifunk-gluon/gluon/commit/898736686382b670b47b323a67cfb36715756fd7 but with the target information pr on top. Don't think that matters.

**Site Configuration:**

https://github.com/freifunk-darmstadt/site-ffda/commit/9e2f01c8d2c5181fdd5d02aae50cf545cd79af6a

**Custom patches:**

None

(Mainly posting this now so that i don't forget and there is some documentation.)

T-X, 2025-05-12:
> But i don't seem them on the node itself on the br-client bridge

I don't think you should see it on br-client. uradvd runs on the local-node interface instead:

```
$ ps | grep radvd
 2142 root      1064 S    /usr/sbin/uradvd -i local-node -p fdef:ffc0:3dd7::/64
```

Also, the according multicast listener for ff02::2 is only on local-node, not br-client:

```
$ ip -o maddr show dev local-node | grep ff02::2
11:	local-node\	inet6 ff02::2
$ ip -o maddr show dev br-client | grep ff02::2
$
```

And with multicast snooping enabled on br-client I would therefore not expect to see the RAs on br-client.

---

Other  than that the VPN connected and unconnected could still be interesting to check. I think RAs had some limit for broadcasted RAs. Wild guess, maybe you have "AdvRASolicitedUnicast off" in your global radvd (even though "on" is recommended by [RFC7772](https://www.rfc-editor.org/rfc/rfc7772.html)), have many incoming, broadcasted RAs from the VPN and that might limit the radvd on the node?

Other things interesting to check: Temporarily disabling multicast snooping. Or setting `echo 2 > /sys/class/net/local-port/brport/multicast_router` to try to enforce a copy of all multicast packets to the local-node interface.

herbetom, 2025-05-13:
> And with multicast snooping enabled on br-client I would therefore not expect to see the RSs + RAs on br-client.

Interesting. My naive assumption was that it should still be visible if it's passing through to another device.

---

What i'm resonably sure about is that i've sometimes got a problem with the `prefix6` at least not staying configured (but sometimes it might also be never configured in the first place idk). I'm noticing this when a node is solely responsible for Advertising the prefix and there is no Gateway to do it also.

I've seen that with multiple Linux PCs with different OSses. (Hoewer both were using for example systemd-networkd (but different versions). So i won't fully rule out a client problem yet.)

rotanid, 2025-11-18:
@herbetom feel free to reopen if the issue persists with current gluon main

closed 2025-11-18 by rotanid