freifunk-gluon/gluon #3520: Warnung: Xiaomi AX3200 Sysupgrade from 23.05 base and earlier to 24.10  fails (and the other direction as well)

ChristianMiddendorf, 2025-06-03:
OpenWRT changes the Flash Layout for Xiaomi AX3200 (AX6)with v24.10.
Actually there is no easy way to Sysupgrade from actual Gluon to the experimental 24.1 based Version.
A downgrade from Gluon based on 24.1 to the stable Series via Sysupgrade fails, too.
See https://openwrt.org/toh/xiaomi/ax3200#upgrading_from_2305_and_earlier_to_upcoming_2410_or_snapshot

The only workaround actually is to restore the Original Firmware and installing the New Gluon from scratch.

Would be great if you find a good Solution . At least the auto updater should not crash the devices.
If crashed… TFTP recovery works.

herbetom, 2025-06-03:
This shouldn't really cause issues with Gluon apart from devices just not updating.

With the autopdater only minor compat version changes are ignored. The process therefore should stop around here: 

https://github.com/freifunk-gluon/packages/blob/bae34a0f53b25dbd691c57c8ab2d0b7cfe811517/admin/autoupdater/src/autoupdater.c#L396

AiyionPrime, 2025-07-17:
@ChristianMiddendorf Lets start with the easy part:
Gluon does not support downgrades.

And for the other part, I think not crashing is the only thing we can offer at this point.
There won't be a feasible way to autoupdate this.

You'd need a two staged update process, where your device loads an initramfs, puts its config somewhere(tm) for safekeeping, then updates to the new firmware with the new flash layout and restores its config from somewhere(tm).

We encountered this before with e.g. the EdgeRouter X, which needs layout changes every so often (in fact it might be due for another one soon?) and with our current stack there is little we can do about it.

As @rotanid put it by adding the respective labels, we should mark this as known issue upon Releasing the first affected firmware. But beside that there's little we can do.

If I misread your issue and the device does in fact crash while updating to the latest master, let me know and I'll investigate further.

AiyionPrime, 2025-07-17:
I hope you don't mind: I updated the title to match the projects scope.

rotanid, 2025-12-31:
as this is mentioned in the Release notes and won't be "fixed" by any means, closing

closed 2025-12-31 by rotanid