freifunk-gluon/gluon #3523: GL.iNet MT3000 is missing bmt-remap-range

Djfe, 2025-06-08:
will extend later

once there is a bad block in the ubi partition, the device will turn into a brick.
recovery won't be able to fix it
for a report see:
https://github.com/openwrt/openwrt/pull/18750#issuecomment-2954153453

blocktrron, 2025-06-12:
@dangowrt Fully aggree, it is stupid and "accident waiting to happen".

Howeder, as `bmt-remap-range` avoids remapping of the area in Linux. If a block goes bad there and we avoid remapping ,the UBI should stay consistent also in the bootloader, as the bad block is only marked as bad on write (which the U-Boot does not do), right?

dangowrt, 2025-06-12:
In case of the MT3000 U-Boot does have the capability to write to UBI as well, eg. when using the HTTP recovery interface to upload a firmware image...

Djfe, 2025-06-12:
Oof, thanks for taking a look. that's sounds more complicated
Sorry, for blaming you for it then @dangowrt 

> They forgot to exclude the ubi partition from nmbm when adding your device, urgh :/

In that case "they" clearly can be redirected towards GL.iNet/Mediatek. I chose "they" because I only looked at the dts at that point not how it came to that point or who committed it, because it didn't matter to me. I wasn't looking to blame anyone actively it just came out wrong.

____________

Anyways, I just looked through all filogic dts files. 29 devices have nmbm activated. 1 of them the Netgear WAX220 uses bmt-remap-range. Though I wonder whether that even works as expected since it also excludes many non ubi partitions in the mtd list that come after the ubi partition for no reason.

Some of the 28 devices have additional variants where you can replace uboot, most don't. Not that it matters since it won't solve the issue, it's rather a workaround.
I wonder how many boards without nmbm in the dts have nmbm enabled within uboot though 😣 (I'm afraid this could affect even more boards in the long run)

dangowrt, 2025-06-12:
It's important to note that `bmt-remap-range` was introduced with the intention to enable the use of UBI (for rootfs and rootfs_data) on devices on which U-Boot uses a BMT or NMBM variant and loads the Linux Kernel from a raw MTD partition. As also the WAX220 loads the kernel from a UBI volume (rather than from a raw MTD partition, like it is common eg. on most MT7621 devices) I think `bmt-remap-range` is not appropriate on the WAX220.

blocktrron, 2025-06-12:
@dangowrt I understand the intent behind `bmt-remap-range`, however i our case excluding the UBI partition from the remap-range would avoid the remapping within UBI.

I fully get the mishap with the factory U-Boot, as remapping would happen on write there too. However this would not change the situation, as avoiding remapping in Linux keeps the regular boot-process in a working state.

In other words, the factory U-Boot recovery would break the partition map itself regardless of the Linux environment, however we can avoid breaking it ourselves (in which case recovery would have to happen from Linux itself).

aiamadeus, 2025-06-13:
> It seems that you are using u-boot to load kernel from UBI on top of nmbm. In this case, you should not set bmt-remap-range and you should let nmbm code handle the bad blocks on the entire flash, otherwise u-boot and kernel will disagree on mtd block layout when any bad block occurs outside this range.

https://github.com/openwrt/openwrt/pull/10690#discussion_r989175214
I think wax220 defined bmt-remap-range incorrectly.

closed 2025-06-13 by blocktrron