freifunk-gluon/gluon #3525: Testreport Gluon-2025 image on AVM 7360v2 - WLAN can not be turned off

fsffunk, 2025-06-15:
### Bug report

**What is the problem?**
When i disable WLAN in settings, the WLAN LED stay on, and WLAN stay on. I tried few times, but it always stay on.

**What is the expected behaviour?**
When i turn off WLAN, it should also turn off the LED and not be visable and usable on WLAN devices. 

**Gluon Version:**
https://firmware.freifunk-aachen.de/firmware/download/from-2025.1.x/sysupgrade/stable.manifest

Djfe, 2025-06-15:
good report!

fsffunk, 2025-06-18:
Is it maybe possible that some other person get assigned to this issue? I have really extremely bad experience with @Djfe . After that terrible experience i stopped doing things with freifunk for a long while.

maurerle, 2025-06-24:
Hello @fsffunk,
thats a weird issue.
By "When i turn off WLAN" you mean through the config-mode of your firmware, unchecking the checkbox for wifi, clicking save button, leaving config mode by finishing..?

If you have ssh access, it would be good to share the output of `uci show wireless` (remove private-wifi password if enabled though) and the output of `iwinfo`.

fsffunk, 2025-06-25:
@maurerle 
Yes, i also had never seen such problem with freifunk before.

> By "When i turn off WLAN" you mean through the config-mode of your firmware, unchecking the checkbox for wifi, clicking save button, leaving config mode by finishing.
And after rebooting, wifi is still turned on?

Exactly.
Before i opened this issue i payed around some bit. For example the offline-ssid worked fine, when i disabled it, saved, rebooted, then it was turned off and the ssid did not changed any more when offline.
Also for example turning off the hidden ssid for mesh. I turned it off, saved, rebootet, it was then gone. 
When i go again into config, the mesh was still disabled. From this 3 WLAN related settings, only the disable of WLAN never worked. I saved, finished config, reboot, WLAN LED got turned on bootup and WLAN functional like all the time before. I then put it again into setting mode, turned it off again(the checkbox was enabled again after reboot), and it was again turned on.
Then i opened this issue.

@Djfe 
Together with cutting contact i also got rid of the matrix account. You can not contact me any more over matrix. If you are sorry, then just finish what was started. I have spend so much time over so many months into this. The only single thing that is relevant is that this have the expected outcome.

Djfe, 2025-06-25:
> @Djfe
Together with cutting contact i also got rid of the matrix account. You can not contact me any more over matrix. If you are sorry, then just finish what was started. I have spend so much time over so many months into this. The only single thing that is relevant is that this have the expected outcome.

sure. This was about adding support for that GL.iNet device, right?
I need the model name again. I lost access to most matrix messages due to encryption in November when I switched smartphones. (the keys weren't transferred correctly 😣)

AiyionPrime, 2025-06-28:
> Is it maybe possible that some other person get assigned to this issue? I have really extremely bad experience with [@Djfe](https://github.com/Djfe) . After that terrible experience i stopped doing things with freifunk for a long while.

Just to make this clear, @Djfe has never been *assigned* to this issue.
He's just a user of gluon like you are.

fsffunk, 2025-06-30:
I could test in between the WLAN-button. Yes, pressing the WLAN-button is working and changes the WLAN from on to off and from off to on when the device is fully boot up.
Configuring the router with the configuration page should be still the main way to configure the device.

maurerle, 2025-07-04:
@fsffunk to gather more information about your issue, you need SSH access or wait for someone else with this device to invest the time to test this.

A introduction to SSH for Freifunk is also [given here](https://wiki.freifunk.net/Konsole#Starten_einer_SSH-Verbindung_unter_dem_Betriebssystem_Linux)

> it would be good to share the output of `uci show wireless` (remove private-wifi password if enabled though) and the output of `iwinfo`.

See here for some more info on this:
https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration#disable-client-network

`uci show wireless` shows the config which should be loaded, `wifi` restarts the radios and applies the uci config, `iwinfo` shows a list of currently active SSIDs.

AiyionPrime, 2025-07-17:
I just built the latest gluon commit with the site from hanover.
`gluon-v2023.2-505-g7b3f7e`, still can't reproduce.

I enter the configmode, is disable wifi client and mesh, the wifi is off.
Everything works as expected.

AiyionPrime, 2025-07-17:
I can **confirm**, Aachens firmware called `v2025.0.0-5`, does indeed no respect the setting.

All standard Aachen configuration, wifi disabled gladly enables wifi.

AiyionPrime, 2025-07-17:
> @maurerle und ich.
> sehen wir uns an

> Ich würd' das dann erstmal dicht machen.

> Ja bitte 👍

 _Originally posted by @Djfe and me in hackint_

@fsffunk Thanks for bringing this up, it will likely enhance a community package.
Until someone shows this is indeed a gluon and not a ffac firmware issue, we'll close this and thank for everybody's attention.

closed 2025-07-17 by AiyionPrime

fsffunk, 2025-12-26:
Thanks. The image i tested now did not had this WLAN issue on an other 7360v2 ( Art. Nr.: 2000 2522 ):

https://firmware.freifunk-aachen.de/firmware/download/from-2025.1.x/sysupgrade/gluon-ffac-v2025.0.0-12-avm-fritz-box-7360-v2-sysupgrade.bin