freifunk-gluon/gluon #3526: Bring back AVM FRITZ!Box 7320 7330 7330SL

fsffunk, 2025-06-15:
Those devices was supported in Gluon in year 2020 but got disabled: https://github.com/freifunk-gluon/gluon/pull/1995

OpenWrt added upstream support for the devices in 2021: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=ceac4ae3b403d52735f371b741f9fd6cbbe08b31

I can confirm that its working fine with OpenWrt. The issue described here https://github.com/freifunk-gluon/gluon/issues/1943 is gone when using the official OpenWrt upstrem image.
The functionality of those boxes for freifunk is the exactly same like for example the AVM 7312 that is supported in gluon and also have just one single LAN port.
At first boot you connect to LAN1 port, configure the device, reboot and use the same port as WAN. If you want to configure it again, hold DECT button until it reboots. Then reconfigure it.

rotanid, 2025-06-15:
@fsffunk please provide evidence for the bug fix, as it is still open in openwrt bug tracker:
https://github.com/openwrt/openwrt/issues/7125
after that, feel free to open a Pull Request that follows the device criteria:
https://github.com/freifunk-gluon/gluon/wiki/Device-Integration
https://github.com/freifunk-gluon/gluon/wiki/Device-Integration-checklist

blocktrron, 2025-06-15:
We don't track issues for devices, new devices are only added by a Pull Request from the person completing the checklist.

closed 2025-06-15 by blocktrron