freifunk-gluon/gluon #3552: Ghost clients on Cudy WR3000 v1

dr2okevin, 2025-07-23:
### Bug report
I observed the following. When clients go out of range, there is a chance that the router keeps them as ghosts. They appear an disappear in the mesh as clients even hours after being physically away. The clients are visible on all typical moitoring like the meshviewer or grafana or node status page. They can appear as 2,4ghz or 5ghz clients. A reboot fixes the issue temporally until the next real clients connects and disappears.
I already tried a completely reset and new firmware flash, but with no differences in the outcome.

logread shows the following (mac addresses shorted my be)
```
Wed Jul 23 14:57:15 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK 40:4e:36:**:**:**
Wed Jul 23 14:59:13 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK cc:73:15:**:**:**
Wed Jul 23 14:59:28 2025 daemon.notice hostapd: client0: AP-STA-POLL-OK ea:42:bb:**:**:**
Wed Jul 23 15:02:32 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK 40:4e:36:**:**:**
Wed Jul 23 15:04:25 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK cc:73:15:**:**:**
Wed Jul 23 15:04:31 2025 daemon.notice hostapd: client0: AP-STA-POLL-OK ea:42:bb:**:**:**
Wed Jul 23 15:07:37 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK 40:4e:36:**:**:**
Wed Jul 23 15:09:32 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK cc:73:15:**:**:**
Wed Jul 23 15:09:45 2025 daemon.notice hostapd: client0: AP-STA-POLL-OK ea:42:bb:**:**:**
Wed Jul 23 15:12:38 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK 40:4e:36:**:**:**
Wed Jul 23 15:14:42 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK cc:73:15:**:**:**
Wed Jul 23 15:15:01 2025 daemon.notice hostapd: client0: AP-STA-POLL-OK ea:42:bb:**:**:**
Wed Jul 23 15:17:56 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK 40:4e:36:**:**:**
Wed Jul 23 15:19:58 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK cc:73:15:**:**:**
Wed Jul 23 15:20:04 2025 daemon.notice hostapd: client0: AP-STA-POLL-OK ea:42:bb:**:**:**
Wed Jul 23 15:23:15 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK 40:4e:36:**:**:**
Wed Jul 23 15:25:13 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK cc:73:15:**:**:**
Wed Jul 23 15:25:22 2025 daemon.notice hostapd: client0: AP-STA-POLL-OK ea:42:bb:**:**:**
Wed Jul 23 15:28:16 2025 daemon.notice hostapd: client1: AP-STA-POLL-OK 40:4e:36:**:**:**
```

Meshviewer link: https://2.meshviewer.pinneberg.freifunk.net/#!/de/map/80afca5cb846
Grafana: https://grafana.pinneberg.freifunk.net/goto/ie3sb3UNR?orgId=1

**Gluon Version:**
2023.2.5

**Site Configuration:**
https://github.com/freifunk-pinneberg/firmware/tree/098f7145abecb831190a9e6f1d81d963136580a5

dr2okevin, 2025-07-27:
I have ordered a new device for testing. Might take 1-2 weeks until I can start test.

AiyionPrime, 2025-09-15:
@dr2okevin If you need help compiling the image for your device, just let us know;
either way we're interested in your results :)

dr2okevin, 2025-11-29:
I will test as soon as I find time. Currently in a lot of stress.

neocturne, 2025-12-25:
@dr2okevin An LLVM toolchain is needed as a build dependency now, see https://gluon.readthedocs.io/en/main/user/getting_started.html#dependencies