freifunk-gluon/gluon #3576: x86_64: order of ethernet ports reversed - regression after upgrade from v2023.2.x to main

Djfe, 2025-09-02:
### Bug report

**What is the problem?**
gluon main reverses the order of ethernet ports on x86 in comparison to the last release `v2023.2.x`
atleast when one ethernet port is internal and one is a pci card. (I haven't tested further cases, yet)

**What is the expected behaviour?**
internal LAN port should stay WAN (uplink role in this case)
pci card should stay LAN (mesh role in this case)

**Gluon Version:**
76f2a3db0da8ee7ae3f800b80cce6f5411084f6d
but with OpenWrt slightly newer:
https://github.com/openwrt/openwrt/commit/03fe5966997363a08d686bca219b647f58ef4fc3
This first occured with an official gluon main build without this kind of patch, too.

**Site Configuration:**
not publicly available, an older state is here:
https://github.com/ffac/site/tree/v2025.1.x

**Custom patches:**
also not the current state:
https://github.com/ffac/site/tree/v2025.1.x/patches

We have to try reproducing this on different machines/VMs now anyways so I might create a separate build without all these changes or even the example site from the gluon repo.

imho this is a regression that needs to be fixed before tagging a new release from gluon main.

Djfe, 2025-09-03:
possibly related/cause?
https://github.com/openwrt/openwrt/commit/8e5c9dbb177ed98562d825a67d62bf30aa6874cf

neocturne, 2025-10-18:
Module lists in (reverse) load order from `/proc/modules`:

[x86-64.main.txt](https://github.com/user-attachments/files/22984077/x86-64.main.txt)
[x86-64.v2023.2.x.txt](https://github.com/user-attachments/files/22984075/x86-64.v2023.2.x.txt)
[x86-generic.main.txt](https://github.com/user-attachments/files/22984074/x86-generic.main.txt)
[x86-generic.v2023.2.x.txt](https://github.com/user-attachments/files/22984076/x86-generic.v2023.2.x.txt)

Looks like on x86-generic it's just r8169 that got moved to the front of a few other drivers, on x86-86 it's r8169 and igc (+ some new drivers that might also affect load order... but let's ignore those).

neocturne, 2025-10-18:
Proposal to prevent such issues in the future, based on @Djfe's idea to use PCI paths to identify devices:

On x86 and similar targets (armvirt?) that don't have clearly defined "LAN" and "WAN" ports, replace the use of the values `/lan`, `/wan` and `/single` for the `name` option in UCI interface sections with something like `/path:pci0000:00/0000:00:01.0`.

This will require some research into the possible sysfs paths on different systems; in particular we must ensure that paths are unique when a single device provides multiple Ethernet interfaces. I believe that hardware using separate PCI functions for multiple interfaces should be fine (they will appear as separate devices), but we should try not to break more "switch-like" devices using a DSA/switchdev or similar driver architectures either.

Maybe the netdev renaming code of systemd/udev could also provide some inspiration?

neocturne, 2025-11-15:
@Djfe My proposed solution would no involve any renaming. It would apply roles configured for PCI paths, regardless of the names of the interfaces.

PCIe paths will not change on reboot. As mentioned before, path based interface naming is what most Linux-distributions use these days - if the names were not stable, this would also break desktop and server systems.

Replacing an Ethernet adapter in the same PCIe slot should retain the same path. I've only seen paths change when PCIe devices were added or removed or PCIe setup in the BIOS was changed; I'd be okay with requiring a manual configuration update in such cases - we don't need to be better than desktop Linux in this regard.

USB devices have paths as well (based on hub port numbers), I don't see a reason to handle them differently. In theory these should be stable across reboots as well.

I don't think we need to handle disappearing paths, this will be very rare. We can show a hint in the UI that the device is not available.

herbetom, 2026-02-07:
What i've done now is included the usb-net modules [we're using at FFRN](https://github.com/Freifunk-Rhein-Neckar/site-ffrn/blob/17ff103dd95e21892e601b9f5d604b777563aa48/image-customization.lua#L86-L102) in the ordering:

https://github.com/herbetom/gluon/commit/783098de2bf6415b8278578db727c9fcda1db0ca

That of course isn't a complete list but i would suspect that it helps in a bunch of cases.

I've already tested it with the x86-64 image on an FUJITSU SIEMENS FUTRO S550 with asix USB-Nic. Without this patch the interfaces are reversed, with it they stay the same.

Is it feasible to just include this upstream? This is a pitfall quite a few will probably encounter.

Just looking at Futros there are [about 427 visible to the census](https://gluon-census.ffrn.de/d/CA5PRFmMz/gluon-census3a-dashboard?orgId=1&from=now-12h&to=now&timezone=browser&var-community=$__all&var-version_major=$__all&var-version=$__all&var-models=FUJITSU%20FUTRO%20S450-2%20&var-models=FUJITSU%20FUTRO%20S520&var-models=FUJITSU%20FUTRO%20S550-2&var-models=FUJITSU%20FUTRO%20S550-2%20&var-models=FUJITSU%20FUTRO%20S700&var-models=FUJITSU%20FUTRO%20S720&var-models=FUJITSU%20FUTRO%20S900&var-models=FUJITSU%20FUTRO%20S920&var-models=FUJITSU%20FUTRO%20S930&var-models=FUJITSU%20SIEMENS%20Futro%20S550%20&var-models=FUJITSU%20SIEMENS%20FUTRO%20S500%20&var-models=FUJITSU%20SIEMENS%20FUTRO%20S550&var-models=FUJITSU%20SIEMENS%20FUTRO%20S550%20&var-models=FUJITSU%20Futro%20550%20&refresh=15m&viewPanel=panel-22). I would suspect quite a few of those are using USB NICs given that it's much cheaper then the necessary PCI-Riser and the PCI NIC.

Btw. the `/proc/modules` diff between v2023.2.4 and v2025.1 is the following:

```diff
--- v2023.2.4   2026-02-07 14:31:58.231665851 +0100
+++ v2025.1     2026-02-07 14:31:58.231665851 +0100
@@ -1,40 +1,40 @@
-# v2023.2.4
+# v2025.1 
 
 smsc95xx
 rndis_host
 mcs7830
 dm9601
 cdc_subset
 cdc_ether
 cdc_eem
 ax88179_178a
 asix
 via_velocity
 via_rhine
 tulip
 sky2
 sis900
-r8169
 r8152
 pegasus
 pcnet32
 ne2k_pci
 ipheth
-igc
 hso
 forcedeth
 e1000e
 e100
 bnx2
 8390
 8139too
 8139cp
 3c59x
 ixgbe
 e1000
 amd_xgbe
 natsemi
 ena
+r8169
 fsl_mph_dr_of
 igb
+igc
 tg3
```

herbetom, 2026-02-07:
Okay, done so: #3685 :)

Just wanted to check there are others who think it's okay to take into account site specific modifiactions in this special case.

---

I've also just verified that x86-generic has the same driver order under v2023.2.x. That seems to be the case. The only difference for me is that the following drivers are missing: `ena`, `amd_xgbe`, `ixgbe` and `igc`. So that should be fine too.

pprindeville, 2026-04-02:
> solution 1 might require this PR to be merged upstream: [openwrt/openwrt#13411](https://github.com/openwrt/openwrt/pull/13411)

If it ever gets reviewed...