freifunk-gluon/gluon #3580: 6 GHz and mesh

maurerle, 2025-09-06:
This is a tracking issue for experience with 6GHz wifi on openwrt/gluon.

The idea is to add 6GHz support after #3563 

First of all:
* 6GHz requires WPA3, so we need to have an OWE interface running.
* Starting an unsecured client interface on the 6ghz radio does simply not come up.
* 6GHz therefore needs OWE without transition mode on the 6ghz wifi
* the private wifi is only available with WPA3
* the mesh channel requires usage of SAE

The question is, which other parts are required for a supported 6GHz mesh radio.

maurerle, 2025-09-06:
```
> cat /var/run/wpa_supplicant-mesh1.conf 
country=DE
network={
	
	ssid="ffac-bat4-mesh"
	key_mgmt=SAE
	mode=5
	fixed_freq=1
	frequency=6115
	ht40=1
	max_oper_chwidth=1
	noscan=1
	sae_password="0c7d4a5e9b703bb3779fe553e02a67a1"
	beacon_int=100
}
```

The problem in the below `iwinfo` is the wrong setup of the mesh1 interface:

```
mesh1     ESSID: "ffac-bat4-mesh"
          Access Point: 9E:6A:C6:09:57:85
          Mode: Mesh Point  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 22 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ax
          Hardware: 14C3:7906 14C3:7906 [MediaTek MT7916AN]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```

<details>
<summary>iwinfo</summary>

```
client0   ESSID: "Freifunk"
          Access Point: 9E:6A:C6:09:57:80
          Mode: Master  Channel: 11 (2.462 GHz)  HT Mode: HE20
          Center Channel 1: 11 2: unknown
          Tx-Power: 20 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -92 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: 14C3:7906 14C3:7906 [MediaTek MT7916AN]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

client2   ESSID: "Freifunk"
          Access Point: 9E:6A:C6:09:57:88
          Mode: Master  Channel: 44 (5.220 GHz)  HT Mode: HE20
          Center Channel 1: 44 2: unknown
          Tx-Power: 23 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -91 dBm
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: embedded [MediaTek MT7986]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy2

mesh0     ESSID: "ffac-bat4-mesh"
          Access Point: 9E:6A:C6:09:57:81
[... 1079 lines cut ...]
	option ssid 'owe.ffac.rocks'

config wifi-iface 'client_radio2'
	option ifname 'client2'
	option network 'client'
	option disabled '0'
	option device 'radio2'
	option mode 'ap'
	option macaddr '9e:6a:c6:09:57:88'
	option ssid 'Freifunk'

config wifi-iface 'wan_radio1'
	option ifname 'wl-wan1'
	option network 'wan'
	option encryption 'sae'
	option device 'radio1'
	option key 'freifunk'
	option disabled '0'
	option mode 'ap'
	option macaddr '9e:6a:c6:09:57:87'
	option ssid 'Private6'
	option ieee80211w '2'
```

</details>

maurerle, 2025-09-07:
Primarly, this is a call for help regarding wifi 6 mesh requirements with logs and a current status.

Most of the information in #2906 is obsolete by now - and stays on a discussion level.

* OWE Client wifi is working without transition mode only
* private wifi works only using WPA3
* mesh always requires SAE (as shown in the logs)
* I don't know if there is a mesh implementation missing for wifi 6, but wpa_supplicants `key_mgmt for mesh network in 6 GHz should be SAE` suggests, that this is implemented as well - maybe there are still some driver support issues

blocktrron, 2025-09-27:
> OWE Client wifi is working without transition mode only

Important to note here is you need to use the Open SSID for the OWE interface on 6GHz and broadcast the SSID.

maurerle, 2025-09-28:
You mean for [RNR](https://wizardfi.com/wifi6e/2024/03/04/reduced-neighbor-report.html) to work?
The owe.freifunk ssid worked with grapheneOS devices as well.
Some other device had issues with the discovery, so this probably must be the same SSID therefore.

maurerle, 2025-09-28:
Regarding mesh - the latest openwrt mt76 update did create a working mesh iface, though we could not test this yet. Will report back, once we did put two 6ghz gluon nodes near each other.

blocktrron, 2025-09-28:
Unrelated to RNR. The configured SSID to broadcast needs to be the open, as otherwise you end up with two SSID for the same network in the scan list.

Transition networks are hidden identified by the matching relation thru the transition IE. This is not available on 6GHz, thus the broadcasted SSID of the OWE network on 6GHz needs to be the Open SSID on a/g band.

maurerle, 2025-10-03:
Thanks @blocktrron - I guess this is not a technical requirement (as it was working before as well), but more from the usability side.
This is now implemented in: https://github.com/ffac/gluon/compare/refactor-wireless-triband...ffac:gluon:6ghz_wifi

As mesh is working now, I will close this issue, so that we only have one 6GHz issue open.
A PR will be created once #3563 is done.
Further discussion of 6GHz can be done in #2906